hathach/tinyusb #249: API shouldn't submit endpoint transfer before enumeration is complete

hathach, 2019-12-27:
**Describe the bug**
Some class driver API is not safe when calling before the SET_CONFIG (enumeration) is complete. For reference https://github.com/adafruit/Adafruit_TinyUSB_ArduinoCore/issues/1#issuecomment-569202207

**Set up (please complete the following information):**
Any setup & examples

pigrew, 2020-01-13:
This seems complicated.

Do you have a proposed fix?

Here are some quick thoughts:

A panic by the core in these cases would be wrong since an interface may be asynchronously deconfigured or reset by the host (such that the application can't predict if the call will be able to succeed or not).

It seems to be that the core need to detect such a mistake, and return an error code which is properly handled by the class and application layers?

Also some sort of synchronization mechanism need to be added for class destruction, to verify that that the app knows that the interface has been destroyed, prior to re-initializing the class at the next set_config?

hathach, 2020-01-14:
@pigrew thank you for lots of great idea, although it is not really complicated one. We only need to check if the endpoint is opened (or probably not zeroes) as it is pointed out with 
https://github.com/hathach/tinyusb/blob/master/src/class/vendor/vendor_device.c#L96
the call to usbd_edpt_busy() may return false with ep = 0 (since it is not opened/configured yet). I will do it later in more systematical way across the class driver later :)