hathach/tinyusb #255: submodule adafruit/asf4 is broken for many SAMD21 parts

majbthrd, 2019-12-28:
tinyusb currently uses commit 82fe3aa05bc3b9f4956ee775c748a3885443f66b from adafruit/asf4 for SAMD files provided in ./hw/mcu/microchip/samd/asf4

The problem with this code is that the include files in ./samd21/include were only selectively hacked up to comment out a #define LITTLE_ENDIAN.  This modification was only done on parts used by Adafruit.  (The ./samd51/include was more consistently modified.)

This precludes using tinyusb as-is on other parts (like a SAMD21J18A or SAMD21E15A, for example).

I would do a pull request with a fix, but subsequent commits on that branch further customize includes to suit just CircuitPython usage, making it even less general purpose.

A tinyusb fork of this code might ensure general purpose usage of tinyusb over time.

hathach, 2020-01-05:
Sorry for late response, I were busy with other works lately. Thank you very much for the issue, Indeed, I also saw the **LITTLE_ENDIAN** redefinition warnings when compiling tinyusb with Arduino as well. However, this is due to asf4 & std C conflict, I think Adafruit try to suppress this but playing it safely. IMHO, The usage of circuitpython samd driver shouldn't prevent tinyusb to be used with other samd parts, since the core of tinyusb (files in `src`) didn't make any use of the mcu driver. It only need register definition and read/write directly to the USB registers. Only minimal set of mcu drivers (clock, gpio & uart) are used by stock example as proof of concept demonstration. 

To be honest, I am not keen on forking & maintaining the samd mcu driver, but I am not holding to it either. Any simple mcu driver that get the stock examples running is fine to me. Later on, if it becomes a real issue, we can change it to un-modified microchip's asf like I did with the samg 

https://github.com/hathach/microchip_driver/tree/master/samg55

majbthrd, 2020-01-07:
I understand not wanting to fork and maintain the samd mcu driver.

closed 2020-01-07 by majbthrd

hathach, 2020-01-07:
thanks @majbthrd , let's wait until it becomes a bigger problem later on :)