hathach/tinyusb #278: Build instructions for examples no longer work

hierophect, 2020-02-05:
**Describe the bug**
Running provided example build instructions:
```
$ cd examples/device/cdc_msc_hid
$ make BOARD=feather_nrf52840_express all
```
no longer results in a successful build:
```
make: realpath: Command not found
/bin/sh: realpath: command not found
../../make.mk:42: *** Invalid BOARD specified.  Stop.
```
**Set up (please complete the following information):**
occurs with any board and any example.

majbthrd, 2020-02-06:
The TinyUSB Makefile depends on (a recent version of) "realpath" being installed.

In some distributions of Linux, opening a terminal and trying to run "realpath" may give recommendations as to which package to install.  Which package seems to vary between Linux distributions.

IMPORTANT: I ran into the same problem, but to make matters worse, my preferred Linux install was a distribution where the only provided version of realpath was 1.19.  This doesn't have the "--relative-to" feature that TinyUSB uses, so the TinyUSB Makefile gave some cryptic error even after I installed realpath.

For what it is worth, on my Linux PC where TinyUSB *will* build, running "realpath --version" reports "GNU coreutils 8.28"; I think the "--relative-to" feature was only added after realpath ceased to be a separate utility and became part of a "coreutils" package.

Hope this helps.

P.S.: if the ./examples/readme.md included information about which software is a prerequisite for building TinyUSB, such situations could be avoided.  Perhaps this could be marked a Suggestion rather than a Bug?

hierophect, 2020-02-06:
Presumably this is only used for examples? I've had no issues using TinyUSB as a submodule of Circuitpython. Additionally, it compiled on my other dev machine (both macs).

majbthrd, 2020-02-06:
Sorry, I don't know the first thing about Circuitpython.  I can only speak to first-hand experience with compiling the examples using the Makefile as a fellow user.

If the Macs (or any other build platform) have already "realpath" installed, it stands to reason that you won't get a "realpath: Command not found" error message.

hierophect, 2020-02-06:
Oh, cripes. I just realized this is a problem with a computer migration I did - I actually had this same issue ages ago and forgot about it. I thought I was having deja vue about this... sorry for the trouble.

closed 2020-02-06 by hierophect

hathach, 2020-02-07:
Thank you @hierophect  for raising the issue, and @majbthrd for answering it, I will find a place to put a notice about this  in the docs