hathach/tinyusb #279: Notes from porting TinyUSB

rajivr, 2020-02-07:
Hi @hathach 

Belated Lunar New Year Greetings and thanks again for TinyUSB. :-)

I managed to get a basic port of TinyUSB working. In our case the DCD layer is in Rust. There are some issues that came up during the port that I wanted to share with you. 

I don't know a good way to create a PR for these changes, so I am providing links to the patches that we currently are maintaining in the tree.

1. We needed to remove `static inline` from `tu_edpt_*` functions. The patch is [here](https://gist.github.com/rajivr/556d63448acf39cd21e8bb97e0d401d3).

    These functions are called from Rust DCD code.

2. We needed to remove `TU_ATTR_WEAK` from a few functions. The patch is [here](https://gist.github.com/rajivr/25ba9904edc93c2e5fd527251f61ca70)

    What was happening in this case was that the linker was picking up the weak definition from TinyUSB instead of the function defined in our crate (Rust's equivalent of a library).

3. Introduce `tud_task_is_queue_empty()` function. The patch is [here](https://gist.github.com/rajivr/2c15302993eceefbfc3104358ffa1a9d)

    We introduced this function because in the OS main loop, we needed some way to determine when TinyUSB did not have any more work left to do, so we can safely execute `wfi` instruction.

4. Because Rust does not have native bitfield support, I had to a workaround by having a `dcd_edpt_open` function written in C that extracted out data from endpoint descriptor before calling Rust DCD function. The code is [here](https://gist.github.com/rajivr/bdbcf473ab288e9505cfe4e5fc19be23)

I hope this is going to be helpful to other TinyUSB/Rust users.

cr1901, 2020-02-08:
I'm not going to profile the static code (mainly because I don't know how :P), but I would rather see a Rust-api specific define (call it `TU_RUST_API`) that [changes](https://github.com/hathach/tinyusb/blob/7b20b80dc0694cfe49b63222f47a8c3d18906e41/src/common/tusb_compiler.h#L67) the definition of `TU_ATTR_WEAK` to nothing, e.g:

```
#ifdef TU_RUST_API
#define TU_ATTR_WEAK      
#else
#define TU_ATTR_WEAK                  __attribute__ ((weak))
#endif
```

And gates whether currently `static` functions remain `static`, e.g. in `tusb_types.h`:

```
#ifndef TU_RUST_API
static inline tusb_dir_t tu_edpt_dir(uint8_t addr)
{
    return (addr & TUSB_DIR_IN_MASK) ? TUSB_DIR_IN : TUSB_DIR_OUT;
}
#endif
```

`tusb_types.c` remains, but is only used by the `cc` crate when compiling `tinyusb-sys` or whatever equivalent you have. If compiling for a C project, `tusb_types.c` is never seen by your project.

hathach, 2020-02-09:
> Hi @hathach
> 
> Belated Lunar New Year Greetings and thanks again for TinyUSB. :-)
> 
> I managed to get a basic port of TinyUSB working. In our case the DCD layer is in Rust. There are some issues that came up during the port that I wanted to share with you.

@rajivr Thank you for your New Year Greetings, I am glad you find TinyUSB useful. 

> I don't know a good way to create a PR for these changes, so I am providing links to the patches that we currently are maintaining in the tree.
>
> 1. We needed to remove `static inline` from [...]

These are helper function, you can simply skip it and define your rust_edpt_* and copy its implementation, those won't be changed anytime soon. 

> 2. We needed to remove `TU_ATTR_WEAK` from a few functions. The patch is [here](https://gist.github.com/rajivr/25ba9904edc93c2e5fd527251f61ca70)
>    What was happening in this case was that the [...]

as @cr1901 point out, you can use `tusb_compiler.h` to overwrite the definition.

> 3. Introduce `tud_task_is_queue_empty()` function. The patch is [here](https://gist.github.com/rajivr/2c15302993eceefbfc3104358ffa1a9d)
>    We introduced this function because in the OS main loop, we [...]

I thought you used Tock RTOS for rust embedded ? most RTOS will automatically to put the caller into suspended mode when queue is empty. For OS NONE configure, tud_task() only return after the event queue is empty. You can sleep after calling tud_task(). 

> 4. Because Rust does not have native bitfield support, I had to a workaround by having a `dcd_edpt_open` function written in C that extracted out data from endpoint descriptor before calling Rust [...]

I don't know Rust, could you just use the bitwise op ( & then shift) to extract the needed field. Sorry if it is dumb question,  I have no idea how it works.

To be honest, I don't see the need to merge any of your proposed changes just yet. Maybe since I don't understand how Rust works.

rajivr, 2020-02-10:
> > 1. Introduce `tud_task_is_queue_empty()` function. The patch is [here](https://gist.github.com/rajivr/2c15302993eceefbfc3104358ffa1a9d)
> >    We introduced this function because in the OS main loop, we needed some way to determine when TinyUSB did not have any more work left to do, so we can safely execute `wfi` instruction.
> 
> I thought you used Tock RTOS for rust embedded ? most RTOS will automatically to put the caller into suspended mode when queue is empty. For OS NONE configure, tud_task() only return after the event queue is empty. You can sleep after calling tud_task().

TockOS, like a lot of other operating systems divides interrupt processing in two halves. The top-half runs in interrupt context, which is mapped to ARMv7-M's handler mode. The bottom-half processing happens in kernel context, which is mapped to ARMv7-M's privileged thread mode. 

The main USB interrupt processing happens in the bottom half, and within the bottom half we add to TinyUSB task queue. Therefore we can call `wfi`, _only when_ both TinyUSB task queue is empty **and** no more bottom half interrupt processing is left to do. 

Hence the need for `tud_task_is_queue_empty()` function.

hathach, 2020-02-11:
> TockOS, like a lot of other operating systems divides interrupt processing in two halves. The top-half runs in interrupt context, which is mapped to ARMv7-M's handler mode. The bottom-half [...]

Tock should implement its own osal_tock port, instead of using osal_none. Like freeRTOS, I would expect the native queue API from RTOS will put the caller into suspended mode if the queue is empty https://github.com/hathach/tinyusb/blob/master/src/osal/osal_freertos.h#L123

cr1901, 2020-04-03:
@rajivr Has there been any forward progress with your TockOS port? I'm looking forward to seeing what you've come up with! :)

hathach, 2020-05-20:
> Introduce tud_task_is_queue_empty() function. The patch is here We introduced this function because in the OS main loop, we needed some way to determine when TinyUSB did not have any more work left to do, so we can safely execute wfi instruction
@rajivr I finally understand your problems, this is new to me, I think your suggestion is correct, and it is indeed needed to safely execute `wfi()`

closed 2020-12-09 by hathach