hathach/tinyusb #281: Set custom device serial number

okyeron, 2020-02-10:
I have been porting some code from Teensy to M0/M4 and for the application I need to set a custom serial number for the device (the serial must be formatted a specific way). 

So far, `setManufacturerDescriptor` and `setProductDescriptor` are available with TinyUSB, but not serial number.

I have hacked to together changes to add ` USBDevice.setSerialDescriptor(serialstr);` mimicking `setProductDescriptor` and it works ok in my limited testing. 

However, in the most recent updates, some "platform dependent codes" have been implemented which includes `getSerialDescriptor()` but I ended up working around that since I don't entirely understand how the "platform dependent codes" are supposed to work

Ideally this should fall back to the current behavior if no custom serial is provided, but I've not figured that out yet.

Since this is somewhat incomplete, I wasn't sure about submitting a PR

What should be my next steps?

hathach, 2020-02-11:
wrong repo, please post your Arduino related port to its own repo here https://github.com/adafruit/Adafruit_TinyUSB_ArduinoCore

okyeron, 2020-02-21:
Thanks for that update. Apologies for posting in the wrong place.

hathach, 2020-02-22:
no problem at all, I initially want to redirect discussion over there. Though I am mostly the maintainer anyway. Currently I am overwhelmed, and haven't got time to look at Arduino port yet. Please be patient.

hathach, 2021-05-19:
duplicated to https://github.com/adafruit/Adafruit_TinyUSB_Arduino/issues/85, closed since this will be implemented there.

closed 2021-05-19 by hathach