hathach/tinyusb #283: Callback for USBDevice.mounted() or similar

PTS93, 2020-02-19:
I was wondering if there is a callback for USBDevice.mounted() or something equivalent that informs me when the device was connected to or disconnected from the host.
I'm writing a library that depends on checking this periodically, would be nice if I can do this without modifying the core. I want to avoid the user having to manually call an update function of the library.

I sifted through the code a bit but couldn't find something obvious.

hathach, 2020-02-26:
@PTS93 there is an callback for  tud_mount_cb/tud_umount_cb in the stack.
https://github.com/hathach/tinyusb/blob/master/src/device/usbd.h#L97
Are you looking for an Arduino callback function ?

PTS93, 2020-02-26:
Yes ultimately I will need it in Arduino. Where would you suggest that I add that setMountedCallback() wrapper for tud_mount_cb/tud_umount_cb? The arduino library itself does not seem to implement any global USB functions but only the specific USB classes. Are those things being done in the Arduino core side?

PTS93, 2020-02-26:
Ah I think I finally understood the setup. So there are 2 Arduino libs essentially. One setting up the general USB and the other is for implementing specific devices like HID or MSD?
So I assume here would be a good place for this callback wrapper?
https://github.com/adafruit/Adafruit_TinyUSB_ArduinoCore/blob/09968245ad675377f2265717f0e138d5dccb39a3/Adafruit_USBD_Device.h

hathach, 2020-02-26:
@PTS93 the core lib implement device + CDC, since cdc is required for Arduino core to perform DFU touch1200 and Serial. The MSC + HID and other higher layers are implemented as wrapper for higher class, which is subject to be changed more frequent. It is kind of complicated, but I don't any better way. 

an setMountedCallback() with an function pointer should be fine, you should also add the setUmountedCallback() as well.

hathach, 2020-08-04:
API is implemented https://github.com/adafruit/Adafruit_TinyUSB_ArduinoCore/blob/master/Adafruit_USBD_Device.h#L75

PTS93, 2020-08-04:
but that is not a callback or did you link to the wrong place?
Sorry I meant to integrate this but got side tracked quite a bit.

hathach, 2020-08-04:
right, I forgot what I have said, and thought it is merely for polling, will add one callback later on

closed 2020-12-09 by hathach