hathach/tinyusb #300: ZLP skipped for zero-length replies to CTRL IN with non-zero wLength

pigrew, 2020-03-15:
While hunting down issues for the STM32FSDEV, I realized that zero-length replies to CTRL IN with non-zero wLength seem to be broken. On the other hand, I don't know if this is actually ever used....

tud_control_xfer uses the following logic:

https://github.com/hathach/tinyusb/blob/a0eb8a26517040850a0e4185aef398183b04ddc7/src/device/usbd_control.c#L111-L124

This assumes that all zero-length data is a status message. In the case that wLength is non-zero, and the class wants a zero-length reply, the data ZLP is never sent. Also, in this case, buffer may be NULL since the length is zero.

I see two solutions: Keep an extra bit of state to keep track of if the data stage has been completed, OR add another parameter to `tud_control_xfer` to tell it if the message is data or status.

hathach, 2020-03-16:
the ZLP is handled internally  here, you are the PR's author for it :D 
 https://github.com/hathach/tinyusb/blob/a0eb8a26517040850a0e4185aef398183b04ddc7/src/device/usbd_control.c#L178

`tud_control_xfer()` is the public API, it is called once by class driver or application to response to the request. The function will break it into smaller 64-byte packet size and **ZLP** if needed as well. On other port such as nrf52840/samd it will response with ZLP just fine.

hathach, 2020-03-16:
ah I see you want class driver to have the option to response with ZLP as `no data available` for a request from host. Currently, I didn't encounter such as request yet. But it is indeed a valid usage for some classes. If this is an issue, we can change the behavior of `tud_control_xfer()` which will always response to DATA phase, if len = 0 then ZLP (if wLength > 0).  Since there is existing `tud_control_status`, driver must use that to response with status which does only response with status. In case https://github.com/hathach/tinyusb/blob/a0eb8a26517040850a0e4185aef398183b04ddc7/src/device/usbd_control.c#L74

pigrew, 2020-03-16:
@hathach 

I'm don't have a use case at the moment, but I found this corner case through the other debugging I was doing related to the stm32 driver. I don't know of a class that uses this sort of reply (or even if it is legal USB).

I'm happy to just close this issue (perhaps adding a documentation comment somewhere), but it does make sense to make the change you suggested (just removing the check for non-zero length).

hathach, 2020-03-16:
> @hathach
>
> I'm don't have a use case at the moment, but I found this corner case through the other debugging I was doing related to the stm32 driver. I don't know of a class that uses this sort of [...]

Yeah, I agreed, let's keep this issue open, I will add an PR and double check the repo. I think I did response status with `tud_control_transfer` in some cases.

> EDIT: And yes... I have bad memory of what I did or didn't do..... I'm running into that problem with working on the stm32 driver, having no clue why certain parts are in it.

We all run into this problem. An symptom of overworking :face_with_head_bandage:

PR #310 (2020-03-28): USBD: Fix zero-length EP0 responses
Closes #300 

Upon reflection, it seems letting tud_control_xfer handle status packets may be a good idea.

For example, the first wLength bytes can be requested as a string descriptor. If `wLength=0` in the request, then only the status packet should be sent.  My previously-proposed solution of forbidding calling tud_control_xfer when `wLength=0` means that this situation must be handled in all control handlers.

On the other hand, putting this logic for `wLength=0` here allows minimal changes to the rest of the code.
closed 2020-03-28 by hathach