hathach/tinyusb #313: Remove details on class implementation from USBD core

pigrew, 2020-03-28:
Whereas I would like to create a new class `usbtest` to use for automated validation of the USBD core + DCD, and currently the USBD core contains the initialization structures of the classes (in `usbd.c`), and that list is getting quite long,

I propose to move to remove class-specific details from the USBD core (as much as possible).

The way most stacks do this is to have the application dynamically register the class implementation during library initialization.

One solution would be for USBD to maintain an array of registered classes (with some limit CFG_TUD_MAX_CLASSES, perhaps defaulting to 4).

An alternative would be for the application to create a `usbd_class_driver_inst_t` structure, which would be a node in a linked list of pointers to `usbd_class_driver_t`.

Thoughts?

hathach, 2020-03-28:
That is a good idea, however I don't really like the dynamic solution for something that doesn't change at all ( set of class drivers). In general, I don't want to use any dynamic solution with the stack. I will rather go to the worst-case scenario to make sure it works every time and prevent run-time error.  

Also I don't like to give application any complicated structure to declare or any chance to pass an incorrect driver struct that could lead to hardfault on the usbd side. Since we don't add class driver that often, this won't really an issue.

pigrew, 2020-03-28:
Ok, I see. Will just press ahead with a `usbtest` module, using the current structure.

closed 2020-03-28 by pigrew

hathach, 2020-03-28:
Yeah, that would be perfect :)