hathach/tinyusb #316: Add ESP32-S2 port

hathach, 2020-03-30:
**Description**
ESP32-S2 is already supported in ESP-IDF https://github.com/espressif/esp-idf/tree/master/components/tinyusb/port/esp32s2 . Although the IDF is specific to ESP32-S2, it is best to integrate it to current bare-metal build of our repo for consistency across ports and helps testing with new features, changes and class driver easier.

Note: since ESP32-S2 use the same Synopsis design ware IP as STM32F4. It makes sense to merge their DCD together if applicable.

cr1901, 2020-03-30:
@hathach Ahhh yes, @ladyada mentioned this part to me a while back. I think their DCDs (not necessarily interrupts, however) should be the same, though it may make sense for the time being to copy the STM32F4 `dcd_synposys.c` to an `ESP32` directory to see how much code can actually be shared after the port is working.

hathach, 2020-03-31:
@cr1901 yeah, that makes sense, we will keep them separated for now. And decide later if needed. Also the USB specs of EPS32S2 isn't available yet, it is better to keep it as it is :)

igrr, 2020-03-31:
At the moment ESP32-S2 port (as included in IDF) uses FIFOs, while the STM port seems to use DMA. IIRC there are some alignment requirements for the USB DMA buffers, and ESP-IDF heap allocator didn't support custom alignment values when we started porting TinyUSB. Alignment support [is implemented now](https://github.com/espressif/esp-idf/commit/d26ddaa644997a8b65dfa52b716e5d9502f69c9e#diff-b43890c7d22cca8d9ae095e020ef36cb), so we could switch to a DMA based as well. This needs some testing, however, so adding the FIFO-based driver first might be a quicker option.

cr1901, 2020-03-31:
@igrr FWIW, the Synopsys driver in tinyusb doesn't use DMA; it's purely interrupt-driven.

igrr, 2020-03-31:
You are right, of course; sorry, I must have been looking at some other code for the ST.

hathach, 2020-04-01:
@cr1901 @igrr yeah, I will just use the dcd as it is for now. We can always improve later :)

hathach, 2020-04-11:
implemented by PR #319 

@igrr Should you be interested on TinyUSB on Arduino. Adafruit also has an Adruino version of TinyUSB. It is composed of 2 parts
- the core inside the BSP which takes cares of CDC (for uploading) https://github.com/adafruit/Adafruit_TinyUSB_ArduinoCore
- and user library https://github.com/adafruit/Adafruit_TinyUSB_Arduino for other class CDC/MSC/MID etc ... 

The core requires a few API port to run with specific mcu, it is mean to be included to BSP, you can look at samd port for the detail here https://github.com/adafruit/ArduinoCore-samd/tree/master/cores/arduino/TinyUSB . Later on I will update the core to match latest version that support esp32s2.

closed 2020-04-11 by hathach