hathach/tinyusb #322: Callback from ISR?

tannewt, 2020-04-03:
Could we add a callback from the `dcd_isr` or indirectly when an event is queued to the buffer? Calling into TinyUSB and other polling tasks repeatedly adds up to a significant overhead. By having TinyUSB call back when queueing a task, we'll know exactly when it needs to be called and do so once we're back on the main thread of execution.

majbthrd, 2020-04-04:
Ideally, it seems undesirable for TinyUSB to callback every time it adds something to the event queue; I think what is needed is a single determination at the end of the ISR as to whether there is anything queued.

Knowing whether TinyUSB has queued anything should be as straightforward as adding something like this to ./src/device/usbd.c

bool usbd_events_are_queued(void)
{
  return !tu_fifo_empty(&_usbd_q->ff);
}

If the last thing done by any dcd_isr() was to call some routine that invoked a check like the above, that routine could then conditionally invoke a callback if events were queued.  There could be a weak declaration of the callback that did nothing for users that didn't need this mechanism.  However, users that did need it could use it to do whatever more efficient mechanism was needed (set a semaphore?) to cause a task to run tud_task().

pigrew, 2020-04-04:
Thanks for pointing out `dcd_isr`. I didn't use it in stm32... now I gotta go update the driver to use it.

The library was intended for use with a RTOS where TinyUSB was a task. The RTOS would handle the waiting on the event queue, so no polling is necessary. With FreeRTOS, `tud_task()` will handle events forever, and never return.

On the other hand, it sounds like you want to be able to wait on a semaphore/event_flags shared by usbtmc and other libraries (for micropython?), and you want to release that shared semaphore in TinyUSB.

The cleanest way might be to create a new "OSAL" header for yourself, so you can intercept the queueing functions. A more hackish way would be to add your callback at the end of the `src/device/usbd.c:dcd_event_handler(...)`. Note that this function can be called from interrupt handlers and also from normal code.

--

Currently the OSAL functions are implemented as inline functions in a `.h` file, and there's no config option to include a custom OSAL. I think that the solution to this issue would be to change the headers so that a custom OSAL could be selected.

In the NO_OS case, `tud_task()` loops over all events until there is no event left. Perhaps it's better for you if it were changed to handling a single event, and then returning true/false based on if it did any work.

tannewt, 2020-04-04:
@pigrew I think `dcd_isr` is just a samd thing. I'm not sure it applies to STM.

Ya, OSAL seems like an interesting hook point. You are spot on with how I want it to set a semaphore. Perhaps it is time to get an RTOS schedule integrated into CircuitPython. So many of the popular RTOSes are monolithic though. Recommendations would be welcome.

I don't think having tud_task returning true or false would help because we're calling it too much now.

pigrew, 2020-04-04:
Dcd_isr is in the dcd header file, so I guess it should be for all ports. I don't know if it would be useful, but the API could be used to (optionally) move interrupt handling into the application thread (instead of in the interrupt context).

If dcd_task only handled one event (instead of all available), perhaps you could switch to a more important task more quickly. You would use the return value to check if all events were handled.

Anyway, lots of possibilities. I don't think there is a clear solution in the codebase at the moment.

hathach, 2020-04-05:
@pigrew @tannewt `dcd_isr()` is work in progress. As discussed previously on other issue, some RTOS as TI-OS or TockOS (not remembered well) requires to call `enterISR()` and `exitISR()` for each irq handler, some system may use different IRQHandler name or even a dynamic ISR installation like mynewt. Therefore I intended to have move IRQHandler() to application, which will call `tud_isr()` (alias to `dcd_isr()`). Though it requires application changes, so I am slowly rolling it out for all ports (but not yet).

For OSAL, yeah, adding an OSAL_CUSTOM makes sense. That will help adding a new RTOS without modifying the stack or want to alternate/testing the current implementation. User need to add include path to `osal_custom.h`. Will make a PR for it.

For `tud_task()`, I did experience an issue with handling a single event, it causes some timeout with a packed of usb events, also it increase chances for osal queue to get overflow with event such as SOF (although it is not used yet).

@tannewt a isr callback is not really ideal and can be poorly used by others that doesn't have a good knowledge of how the stack is functioning. How's about giving circuitpython the IRQHandler() as mentioned above, you could set any flags there if needed

```C
// USB_Handler is moved from TinyUSB to Circuitpython
void USB_Handler(void) 
{
  dcd_isr(0);
  // set circuitpython flag/semaphore
}
``` 

For `tud_task_is_empty()`, @rajivr also suggests this API for TockOS. Though I haven't used TockOS, so I asked him to check out the osal port for TockOS.

Indeed there is a few approaches, maybe we can try a few out before deciding.

pigrew, 2020-04-05:
> Therefore I intended to have move IRQHandler() to application, which will call tud_isr() (alias to dcd_isr()). Though it requires application changes, so I am slowly rolling it out for all ports (but not yet).

I like this feature. Do we need to add an `in_isr` parameter to `dcd_isr()`?

The enable/disable interrupt functions might need to move to the application code, too.

> a isr callback is not really ideal and can be poorly used by others that doesn't have a good knowledge of how the stack is functioning.

Even in this situation with micropython, it makes more sense to have the callback from the enqueue function. Some interrupts might not queue any events, or a class driver could (potentially) queue an event.

-- 
I'm also imagining a case where the interrupt handler is called from the application itself, in order to reduce latency of other interrupts:
```
// USB_Handler is moved from TinyUSB to app
void USB_Handler(void) 
{
  disable_usb_interrupt()
  event_set(event_irq_bit)
}
void task() {
  while(true) {
    event_wait();
    if(event_queue_bit)
      event_clear_bits(event_queue_bit)
      tusb_task()
    if(event_irq_bit)
      event_clear_bits(event_irq_bit)
      dcd_isr(0);
      enable_usb_interrupt();
  }
}
```

tannewt, 2020-04-05:
> @tannewt a isr callback is not really ideal and can be poorly used by others that doesn't have a good knowledge of how the stack is functioning. How's about giving circuitpython the IRQHandler() as [...]

This would work for me! I'm ok triggering tud_task after every interrupt even though it may not have anything to handle. It's better than calling it much more than that.

hathach, 2020-04-05:
@pigrew @tannewt since both of you like the `dcd_isr()` approach, I will make a PR for it tomorrow morning. For the name, maybe make it `dcd_irq_handler()` and (`tud_irq_handler() application alias) for a clearer meaning. Most of the ISR event ended in event queue task (except for a few cases), since the stack try to runs as minimal code as possible in isr context. That is like 95% of the time, so I guess it should be fine.

hathach, 2020-04-06:
> I like this feature. Do we need to add an in_isr parameter to dcd_isr()?

good question, hmm, I think we don't need to have `in_isr` parameter. Application doesn't need to defer dcd_isr(), it is mostly be in isr context.

hathach, 2020-04-13:
resolved by #325. Sum up: tinyusb does not provide ISR callback to application, it now works in reverse, application call `dcd_irq_handler()` in its USB ISR handler.

closed 2020-04-13 by hathach