hathach/tinyusb #340: add dcd_edpt_close() API

hathach, 2020-04-13:
Related to #326, #336, #320 #317 

check list for adding new dcd API to close an endpoint.  This API is needed for supporting 
- multiple configuration and 
- Alternate Interface. 

Although the API will only be used by above usecase, it is probably simple enough to be implemented for all port. Port checklikst

- [ ] Espressif ESP32-S2
- [ ] Microchip SAMD
- [ ] Microchip SAMG 
- [ ] Nordic nRF5x 
- [ ] Nuvoton NUC120
- [ ] Nuvoton NUC121/125/126
- [ ] Nuvoton NUC505
- [ ] NXP LPC3511
- [ ] NXP LPC17/40
- [ ] NXP Transdimension (18/ iMX RT)
- [ ] Sony CXD56
- [ ] STM32 FSDev
- [ ] STM32 Synopsys
- [ ] TI MSP430
- [ ] ValentyUSB eptri (fomu)

JeffGiant, 2020-10-22:
@hathach Do you have plan to support this API for ESP32-S2? Or any update.

hathach, 2020-10-22:
no update and no plan, I will work on it when I got some time. Any PRs is welcome.