hathach/tinyusb #346: usbd_edpt_stall should clear pending transfers on EP

pigrew, 2020-04-15:
usbd_edpt_stall() needs to clear pending transfers from the event queue on the particular EP.

A proposed fix would be to add a call to the queue filtering code in #336.

hathach, 2020-04-15:
I don't know if this is really needed, why should we remove an successful transfer on the queue. Also the stack doesn't queue 2 transfer on the same endpoints, it probably couldn't run into this issue.

pigrew, 2020-04-15:
USBTMC uses stall to cancel a transaction in progress.

USBTMC first uses a control request to cancel the transaction (which will put the EP into stall). It then will issue a second request to check if the cancellation of the transaction is complete. Currently, there's no way for the USBTMC class driver to know if there is still any data in the event queue, or not.

I believe that STALL would only be used in the event that the host wants the device to purge its buffers, so losing data is desired.

hathach, 2020-04-16:
> USBTMC uses stall to cancel a transaction in progress.
> 
> USBTMC first uses a control request to cancel the transaction (which will put the bulk EP into stall). It then will issue a second request to check if the cancellation of the transaction is complete. Currently, there's no way for the USBTMC class driver to know if there is still any data in the event queue. The class driver can't know if cancellation is complete.

You only need to cancel the currently queue transferred and not transferring, the transfer that is already complete in the event queue should still be processed.  
 
> I believe that STALL would only be used in the event that the host wants the device to purge its buffers, so losing data is desired.

It is not quite like that, STALL mean halting response for an endpoints, until some action is taken. The cause is related to what is currently happened than to what is already happened (in event queue). It really depends on the class driver, for MSC when host sending out an unsupported SCSI command, it will STALL one of the endpoints (depending on the direction), host will use control to see what is the reason for the STALL (not supported command, not supported argument, or invalid state etc ...) then send CLEAR_STALL and resume to read the SCSI status. 

[image]

hathach, 2020-04-16:
In general, I don't think it is a good idea to discard any successful event occurs in the past because of the current situation.

hathach, 2026-10-07:
New evidence from https://github.com/hathach/tinyusb/pull/4146 (#4038), RA6M5 USBHS (`rusb2`):

A transfer completion that is already queued when the host's SET_FEATURE(ENDPOINT_HALT) is processed lets a class arm the halted endpoint:

1. `usbd_edpt_stall()` sets STALLED|BUSY.
2. The queued XFER_COMPLETE is processed afterwards and clears BUSY|CLAIMED (`src/device/usbd.c:784-788`; the ISR path at `:1500-1509` does the same).
3. The class re-arms from `xfer_cb` (vendor pump, streams), so `dcd_edpt_xfer()` reaches a halted endpoint while `usbd_edpt_stalled()` still reports it halted.

On rusb2 the re-arm cleared the hardware STALL: usbtest case 13 (set/clear halt) failed 7-14 of 40 runs on ra6m5_ek with `verify_halted failed ... status 0 (not -32)`. The PR makes the DCD keep the halt (an OUT submission is recorded and armed at clear-halt, an IN submission is dropped), giving 0 of 40, but every DCD receives `dcd_edpt_xfer()` on a halted endpoint in this window. A forced fixture (stall leaving BUSY|CLAIMED clear) showed 1280 such submissions per direction in 20 case-13 runs on ra6m5_ek.

Also visible: after clear-halt the DCD-side semantics differ by direction (a halt aborts an IN transfer but keeps an OUT receive queued), while `usbd_edpt_clear_stall()` releases both.

Option beyond discarding queued events: keep BUSY on a completion processed while STALLED (clear only CLAIMED), so claim-based re-arms fail quietly until clear-halt. Needs an audit of classes that re-arm in `xfer_cb` without a claim (msc, mtp, usbtmc would hit `TU_ASSERT(!BUSY)` in `usbd_edpt_xfer`) and of the clear-halt recovery paths. Related: #3962.

hathach, 2026-10-09:
#4146 adds halted-pipe guards in `process_pipe_xfer` (`src/portable/renesas/rusb2/dcd_rusb2.c`) as a DCD-side stopgap for this contract: an IN submit on a halted pipe is dropped, an OUT submit stays queued for `dcd_edpt_clear_stall`. They are candidates for removal once usbd keeps BUSY while STALLED and the rusb2 halt cases (usbtest 13 and 29) are revalidated.