hathach/tinyusb #358: MIDI: missing flow-control drops packets

kaysievers, 2020-04-19:
I'm running this patch to make the tinyusb MIDI code behave as expected:
  https://github.com/versioduo/Adafruit_TinyUSB_ArduinoCore/commit/4e8e4a3d6dcad2313169461bb4b60f8025c21c4e

The git commit of the patch message should explain what is going on. The current tinyusb  MIDI code does not work correctly for anything non-trivial, it just drops random packets. Large sysex transfers (16kb) drop almost half of the packets.

It is more a general USB problem, nothing really MIDI specific. The patch does not apply directly anymore though, there are too  many changes in the meantime. Could please let me know how we can get this fixed?

Should I try to forward-port it, or can you add the proper flow control to the core code?

cr1901, 2020-04-19:
>The patch does not apply directly anymore though, there are too many changes in the meantime.

What happens when you apply that patch as-is to the current HEAD? That was Feb 28? Most activity on tinyusb since then has been dedicated to:

* The ESP32 port
* Getting msp430 merged after it stagnated
* CDC-EEM/ECM support
  * Discussing multiple configuration support
  * Discussing alternate interface support
* Various bug fixes (incl ZLP handling)

Of these, only the bug fix code could really affect MIDI, and I don't think anything really fundamental changed about tinyusb's event-handling code in the past month that would make your patch obsolete.

hathach, 2020-04-19:
> I'm running this patch to make the tinyusb MIDI code behave as expected:
> [versioduo/Adafruit_TinyUSB_ArduinoCore@4e8e4a3](https://github.com/versioduo/Adafruit_TinyUSB_ArduinoCore/commit/4e8e4a3d6dcad2313169461bb4b60f8025c21c4e)
> 
> The git commit of the patch message should explain what is going on. The current tinyusb MIDI code does not work correctly for anything non-trivial, it just drops random packets. Large sysex transfers (16kb) drop almost half of the packets.
> 
> It is more a general USB problem, nothing really MIDI specific. The patch does not apply directly anymore though, there are too many changes in the meantime. Could please let me know how we can get this fixed?
> 
> Should I try to forward-port it, or can you add the proper flow control to the core code?

Yeah, It seems to take some inspiration from cdc. The config fifo for not allow overwriting would help. @kaysievers Can you tell us steps to reproduce this issue as detail as possible (I don't even know how to send sysEx message. ). I will try to analyze on my side.

kaysievers, 2020-04-19:
To reproduce just send as many MIDI packets as possible (what the host will just do).
Run a loop with a known number of packets (to emulate a large sysex message).

The current code is not able to receive them. If you count them on the receiving side, you see 10-50% of the packets are lost.

hathach, 2020-04-19:
> To reproduce just send as many MIDI packets as possible (what the host will just do).
> Run a loop with a known number of packets (to emulate a large sysex message).

OK, sound easy enough, but I am totally noob to this. Could you tell how to `send as many MIDI packets as possible` i.e which app should I use (I am on ubuntu, but I have windows laptop as well). And also I have no idea to `Run a loop with a know number of packets` as well. I think you will probably need to give me firmware code, or something that I could try out.

> The current code is not able to receive them. If you count them on the receiving side, you see 10-50% of the packets are lost.

OK, I also don't know how to tell if packet is lost !!! lol, I am totally noob, never do anything with MIDI except for the midi_test example (which I also got help as well).

kaysievers, 2020-04-19:
I created a minimal test, and reproduced it. Here is a repository with an Arduino sketch:
https://github.com/kaysievers/midi-test

I just copied my MIDI lib into the directory, it should not require anything else. The library assembles kilobytes large sysex messages from the 4 byte packets.

Instructions to run the test on Linux are in:
https://github.com/kaysievers/midi-test/blob/master/test-log.txt

It needs the recently merged tinyusb patch and:
https://github.com/adafruit/Adafruit_TinyUSB_Arduino/pull/55

hathach, 2020-04-21:
fixed by #359 , this issue is fixed is right about the time I do sync up work for Arduino library. Couldn't be better timing.

closed 2020-04-21 by hathach