hathach/tinyusb #360: STM32F103 Hardware requirements

gimbas, 2020-04-19:
**Question Description**
I have a board with the same microcontroller as the bluepill, i compiled an example for the bluepill and it works fine on it, but when i port it to my board it doesn't seem to connect, just blinks a BLINK_NOT_MOUNTED pattern, i still have things i want to test, but i'm asking this now so i can work while witing for a response.

The cystals are not the same, the blue pill has a 8MHz crystal, while my board has a 12MHz, i took that into consideration and it seems to be well configured.

My board has a slightly different connection to the usb pins, which i believe might be the reason for the issue, hence the question being hardware requirements, the blue pill has series resistance on both lines and a pull down on the D+, while mine doesn't have the pull down and has some components in series and a couple capacitors on the data lines, which i believe is just some filtering components and shouldn't really matter.

anyone know what might be the issue? or any clues that might help know what's going on.

gimbas, 2020-04-19:
It is in fact the lack of pull up, i know this board enumerated usb without it on the original firmware, i will try use the internal pull up.

closed 2020-04-19 by gimbas

pigrew, 2020-04-19:
@gimbas,

It should work.

The 1.5k pull up is required. You may want it to be controlled by a GPIO line if the board needs to be able to disconnect itself from the USB bus. If it is self-powered, using a pin to detect Vbus is nice, too.

The internal pull-up will not be reliable, and may even create ESD protection issues.

I forget details on series resistors, but don't use above 27 ohm. I think I've seen 22 used. Capacitors can help somewhat with EMI and ESD protection, but will limit the length of the USB cable. Maybe 47 pF max?

hathach, 2020-04-20:
@gimbas If possible you should try with the Bluepill board, I tested the same mcu there. It may give you an idea.