hathach/tinyusb #362: STM32 does not enumerate without external pull up on DP

gimbas, 2020-04-19:
**Set up (mandatory):**
I have a board that is "identical" to a bluepill, same MCU with less memory, 12MHz crystal instead of 8, but the diference that matters is the lack of pull up on the DP line.
Provide details of your setup help us to reproduce the issue as quick as possible  
 - **PC OS**   : Arch linux 5.5.9
 - **Board**   : Bluepill (Without pull up on DP)
 - **Firmware**: examples/device/hid_composite

**Bug Description**
Without the external pull up resistor on the DP line, the device does not connect/enumerate.
i know for a fact it's possible to use USB on this MCU without the external pull up, since the original firmware works without it, why is the reason for it's existence and is there a workaround?

**Reproduce**
Steps to reproduce the behavior:
1. Remove R10 from a bluepill (it's the pull up on DP)
2. Compile and upload examples/device/hid_composite firmware

pigrew, 2020-04-19:
I replied to #360, but will also reply here.

The internal pull-up is not the correct resistance and will not meet USB specs. I think it's between 50k and 150kohm, whereas USB requires around 1.5k.

The F103 does not have an internal USB-compatible pull-up. Other models like F072 do.

This cannot be changed/fixed on a F103, it's a limitation of the hardware.

Read ST AN4879 for more details.

gimbas, 2020-04-19:
How is it that the original firmware works then? this is an of the shelf product, that implements an USB HID device on a STM32F103 and it does not have the pull-up by default, there must be a work around.

gimbas, 2020-04-19:
just looked at the other issue, you said it should work with the internal pull-up? it did not when i tried it.

gimbas, 2020-04-19:
I should mention, this is not my hardware, i am working on an open-source alternative to the original firmware, so modifying the hardware is not really an option

gimbas, 2020-04-19:
Nevermind, i overlooked something while reverse engineering the board, there is in fact a pull-up connected to a gpio.

sorry for bothering you, and thank you.

closed 2020-04-19 by gimbas

pigrew, 2020-04-19:
I was meaning that it would work with a 1.5k added... 

It won't work (reliably) without an external pull-up. Maybe it would work sometimes with the internal pull-up (I've not tried).

Do you have a board photo? Could it have a pull up hidden somewhere (like in an ESD protection IC)?

pigrew, 2020-04-19:
Sounds good then... We just added some code (dcd_connect) which you can use in the BSP source file to enable the IO pin