hathach/tinyusb #366: HID USB with 16bit deltas

gimbas, 2020-04-21:
**Question Description**
I implemented a usb device that advertises as an hid mouse, it is working great, but i wanted to make a small change, i want to send X & Y delta values in int16 variables, i believe this is possible, but i couldn't get it to work.

am i correct to assume:
I can just make an alternate version of TUD_HID_REPORT_DESC_MOUSE with
```
HID_LOGICAL_MIN_N ( 0x8000, 2                             ) ,\
HID_LOGICAL_MAX_N ( 0x7FFF, 2                           ) ,\
HID_REPORT_COUNT( 2                                  ) ,\
HID_REPORT_SIZE ( 16                                    ) ,\
```

and then a modified ```hid_mouse_report_t```
and
```tud_hid_report(REPORT_ID_MOUSE, &report, sizeof(report));```

what am i missing? i tried it but it stopped enumerating. i didn't want to keep messing around, i'm sure it's straight forward.

hathach, 2020-04-22:
are you able to get any error message from host and/or check your descriptor with other hid parser tool ?

gimbas, 2020-04-22:
the device does not show up on lsusb, i'm not sure how to debug usb devices, is there a log somewhere? I'm running linux (arch) btw.

gimbas, 2020-04-22:
are there tools you recomend?

hathach, 2020-04-25:
failed to enumerate is probably other issue, are you running the stock hid example with only the hid descriptor as modification. For tool, you should buy an hardware usb analyzer, software analyzer is not trustworthy

gimbas, 2020-04-25:
I was running the stock hid composite, it was working fine, those were the only modifications

pigrew, 2020-04-26:
I just tried implementing this, and it seems to work for me. One note is that I had issues after making changes because the makefiles don't properly handle header file dependencies.  I had to `make clean` and `make` again.

See my branch (which I'll probably delete once this issue is closed):

https://github.com/pigrew/tinyusb/tree/16_bit_mouse_hid

I'm using Windows 10.

gimbas, 2020-04-28:
Got it working, the issue lied in that i included ```tinyusb``` as a submodule, so i had to keep changes only to my side of the code, when i declared the alternate report descriptor i had missing includes, probably causing a misconstrued report descriptor.

Thank you for the help.

closed 2020-04-28 by gimbas