hathach/tinyusb #377: MIDI data buffer full

hathach, 2020-04-23:
**Set up**
 - **PC OS**   : Ubuntu 18.04
 - **Board**   : Feather nRF52840 Express
 - **Firmware**: examples/device/midi_test

**Describe the bug**
PR #359 fixes the issue with sysex by changing FIFO from overwritable to not, which is fine with sysex message but not OK when streaming data. 

**To Reproduce**
Steps to reproduce the behavior using MIDI-OX
1. Go to 'Ooptions->MIDI Device'
2. Select TInyusb for both Input and Output
3. Before PR #359 Note are sent fine
4. After PR #359 MIDI-OX will hang after the first 2 notes, may issue an error as well

**Log & screenshots**
Setting
[image]

Good one
[image]

Bad one
[image]

hathach, 2020-04-23:
@kaysievers Do you know the issue and quick fix, else I am afraid we have to revert the changes and release the BSP for both SAMD and nRF

kaysievers, 2020-04-24:
It is not an acceptable behavior for MIDI to drop data, it has the same rules as CDC, it needs to block the writer not silently discard packets.

The FIFOs must overwrite themselves. I suggest you turn that around and fix the underlying core issues first, before we try to re-apply the patches on top.

hathach, 2020-04-24:
> It is not an acceptable behavior for MIDI to drop data, it has the same rules as CDC, it needs to block the writer not silently discard packets. The FIFOs must not overwrite themselves.
> 
> I suggest you turn that around and fix the underlying core issues first, before we try to re-apply the patches on top.

don't get me wrong, I will re-apply the patch, but I need to buy time to test more with MIDI. Reverting  the patch now to make sure it doesn't introduce new issue to other projects such as Arduino and circuitpython.

kaysievers, 2020-04-24:
Sure, I just meant that we should fix the underlying problem first, by making the current MIDI code not lose its packets or overflow its buffers. This is unrelated to the patches on top.

The current code just does not work for tools that sends packets faster than the test program.

hathach, 2020-04-24:
> Sure, I just meant that we should fix the underlying problem first, by making the current MIDI code not lose its packets or overflow its buffers. This is unrelated to the patches on top.
> 
> The current code just does not work correctly for tools that send packets faster than the test program.

Could you help to modify the current midi_test example to reproduce the same issue that we got with Arduino. That will make it easier to troubleshoot. I mean the sysex printing message count example. I will make sure both that test and this test with MIDI-OX will work.

kaysievers, 2020-09-23:
I've rebased my repositories today for the first time since April. Everything seems to work fine. There is only one patch left.

Here is the current change to the MIDI interface to make it work reliably with SysEx messages:
  https://github.com/versioduo/Adafruit_TinyUSB_ArduinoCore/commits/versioduo

hathach, 2020-09-25:
@kaysievers Thanks for letting me know, there is a couple of bug fix regarding MIDI later in the original stack recently.
https://github.com/hathach/tinyusb/pull/512
https://github.com/hathach/tinyusb/pull/497

I still haven't got time to check the PR last time we reverted yet. Will do it later on when having time, it probably works better now.

kaysievers, 2020-09-28:
>  it probably works better now.

The FIFOs still drop data, right? As long as that is the case, the current MIDI code cannot reliably work. MIDI packet flow needs to behave like a CDC serial line, just block, never drop anything.

Could you fix the MIDI FIFOs to work as expected?

hathach, 2020-09-30:
> > it probably works better now.
> 
> The FIFOs still drop data, right? As long as that is the case, the current MIDI code cannot reliably work. MIDI packet flow needs to behave like a CDC serial line, just block, never drop anything.
> 
> Could you fix the MIDI FIFOs to work as expected?

The fifo is still overwritable for MIDI, I still haven't got time too look at this issue yet since the last time. I will try to do it later on.

hathach, 2021-02-19:
closed by #618

closed 2021-02-19 by hathach