hathach/tinyusb #381: Prevent register race condition for DCD Synopsys

hathach, 2020-04-27:
**Is your feature request related to a problem? Please describe.**
Several register of dcd synopsys e.g DIEPEMPMSK,  are read-modify-write in both normal & ISR context which can have a race condition

**Describe the solution you'd like**
use int disable/enable to prevent conflict.

https://github.com/hathach/tinyusb/pull/380#discussion_r415410615

liangxiaoLv, 2026-09-28:
We hit this on ESP32-S3 (ESP-IDF 6.0, dwc2 slave mode) and instrumented the
`DIEPEMPMSK` writes to catch it in the act: the losing write read back the whole
word cleared, taking the other endpoint's bit with it. The victim was a bulk IN
endpoint, which then latched permanently - `DIEPTSIZ.XferSize > 0`, `DTXFSTS` = 0
(FIFO empty), `DIEPEMPMSK` bit clear - while the isochronous IN endpoint kept
working (it re-arms every frame and self-heals).

Rate and signature: 4 events / 6120 s over 20x120 s + 2x3600 s runs; 0 / 9600 s
(~2.0e8 mask writes) after serializing the two RMW sites.

PR #4013 proposes the fix for the `DIEPEMPMSK` case: both sites (the slave-mode
set in `edpt_schedule_packets()`, the clear in `handle_epin_slave()`) wrapped in
the OSAL spinlock that `usbd.c` already uses, so there is no platform-specific
code. #3940 covers the bus-reset case, which is a different path into a stale
mask.

Happy to adjust the approach if you would rather have the interrupts disabled
directly (`in_isr` aware) or see these registers handled some other way.

liangxiaoLv, 2026-09-28:
Scope of the issue: `DIEPEMPMSK` is not the only register in this state. We
audited the ones that are read-modify-written from more than one context on
master `3024dd2e3` (`dcd_dwc2.c` line numbers):

- `gintmsk` - the ISR toggles `USBSUSPM` / `OTGINT` / `SOFM` / `RXFLVLM` inside
  `dcd_int_handler()` (1255, 1261, 1267, 1279, 1288, 1293, 1309, 1315), while the
  task side writes the same word in `dcd_sof_enable()` (606, 608),
  `dcd_remote_wakeup()` (549) and `dcd_init()` (507). The file itself notes that
  audio/video classes use `dcd_sof_enable()` on purpose (617), so a lost
  `GINTMSK_SOFM` there costs the class its frame timing.
- `dctl` - `edpt_disable()` read-modify-writes `SGONAK`/`CGONAK` (354, 364) and
  runs from the ISR (`handle_rxvlvl_irq` 950, `handle_epout_dma` 1048,
  `handle_incomplete_iso_in` 1216) as well as from the task (`dcd_edpt_stall`
  731, `dcd_edpt_iso_activate` 660); the task also writes `RWUSIG` (545, 554)
  and `SDIS` (574, 594).
- `daintmsk` - `edpt_activate()` does `|=` from the task (314) with no lock held,
  while `handle_bus_reset()` writes the whole word from the ISR (784).
- per-endpoint `diepctl`/`doeptsiz` - the same-endpoint task-vs-ISR window, i.e.
  the ISR acting on a transfer the task is re-arming at that moment. That is the
  same class as the `epin->tsiz` read we had to pull inside the lock in #4013,
  but a different pattern (same register and endpoint, not different bits of one
  word).

These are source-read candidates, not something we have tried to trigger; each
needs its own reachability check. #4013 is deliberately limited to the
`DIEPEMPMSK` word where we could watch a bit disappear.

On the solution you asked for, "use int disable/enable": that is what
`usbd_spin_lock(false)` does on a single-core OSAL - `osal_none.h` maps it to
`interrupt_set(false)`, which is `dcd_int_disable()`. On an SMP part it is not
sufficient, because masking interrupts only stops the ISR on the local core and
the USB IRQ can be running on the other one; that is how the bit was lost on
ESP32-S3. So #4013 uses the OSAL spinlock, which is a cross-core critical section
on ESP-IDF's portMUX and reduces to int disable/enable everywhere else (Pico,
Zephyr, RT-Thread, ThreadX). If you would rather have it written as a direct
disable/enable somewhere specific, say where and we will change it.

One consequence worth knowing for #3940: it clears `diepempmsk` on bus reset,
and as long as the clear path is not serialized that store can be undone by a
concurrent stale clear. With the clear now under the same lock as the set and as
`handle_bus_reset()`, all three writers are in one protection domain.

hathach, 2026-10-08:
`DIEPEMPMSK` is serialized by #4013 and `GINTMSK`/`DAINTMSK` by #4115, both with `usbd_spin_lock`. The remaining register with task-vs-ISR read-modify-writes is `DCTL` (task: remote wakeup, connect/disconnect, OUT stall, test mode; ISR: `pti_configure()` in bus reset), tracked in #4157.

liangxiaoLv, 2026-10-13:
Thanks for the follow-up and for tracking the remaining register paths. Glad that #4013 covers the DIEPEMPMSK case; I’ll keep an eye on #4115 and #4157.