hathach/tinyusb #382: Merge DCD for stm32 synopsys and esp32s2

hathach, 2020-04-27:
**Is your feature request related to a problem? Please describe.**
95% of stm32 synopsys and esp32s2 are the same. The differences are mostly register name and address base. Merging them reduce maintaining effort for driver. It is also a good chance to generalize the synopsys driver making adding MCU from other vendor using the same IP easier. 

**Describe the solution you'd like**
merge and rename common driver as `synopsys/dcd_synopsys`

hathach, 2021-01-22:
closed seem it doesn't seem to be feasible at the moment

cr1901, 2021-02-21:
Yea, I'm a month late.

I agree this is the best approach for now, but I'd love to eventually see if a "generic" driver is possible. Or at least whether there are "common" parts that are unlikely to change that can be factored out  (_or_ parts that change the same way for all chips using Synopsys IP as tinyusb evolves) .

hathach, 2021-02-22:
@cr1901 yeah, you are right, maybe we could extract the hanlding/processing function such as transfer into common function shared by these dcd rather than merging them all together. I will re-open this and will look at this later on.

cr1901, 2021-02-22:
@hathach Appreciated, especially since I was a month late to actually _say_ anything :D!

hathach, 2021-02-22:
no problem at all, I mostly file issue whenever idea come up as reminder anyway :)

hathach, 2021-03-05:
Seem like EFM from EnergyMicro/Silab also use Synopsys as well, change The issue name for betting meaning. I am hoping to implement it like ehci/ohci where multiple vendors can calls/implement common API to get the driver running.

cr1901, 2021-03-05:
If you're not using a broken compiler, a common API will be zero runtime cost anyway thanks to LTO. So this seems like a good idea.

Novakov, 2021-03-27:
GD32VF103 (RISC-V from GigaDevice) is also using the same driver, in fact I was able to run TinyUSB just by copying stm32f1_fsdev driver plus some structs/defines from stm32 libraries - no changes at all :)

hathach, 2021-03-29:
thanks for confirmation that the stack work with gd32vf103, I also have this board in my drawer as well. Will put it into the official board later on when having time. However, fsdev driver is totally different than the synopsys one mentioned in this issue.

Novakov, 2021-03-29:
> thanks for confirmation that the stack work with gd32vf103, I also have this board in my drawer as well. Will put it into the official board later on when having time. However, fsdev driver is totally different than the synopsys one mentioned in this issue.

Sic, it seems that my reading skills are failing me. Sorry for making mess. Should I remove my comment?

hathach, 2021-03-29:
no problems at all, I have no issues with your comments. At least it provide information that it does work with the gd f103 risc-v :)

hathach, 2021-08-10:
@Novakov I am sorry for mistaking gdvf103 (rsicv) for the gd32f103 (m3). Apparently you are spot on, the risc v version use synopsys driver there is on-going PR to add support for vf103 that you may be interested in #959

PR #1163 (2021-11-04): Generalize synopsys dwc2
**Describe the PR**
Generalize Synopsys DWC2 driver. New driver is at `synopsys/dwc2` For the time being, application can still pick the existing driver e.g st/synopsys/dcd_synopsys.c for stm32. Using this generalized driver is user choice until proven reliably. 

MCUs that use DWC2
- ST STM32 (F4, F7, H7 etc ..) OTG
- Espressif ESP32 S2/S3
- Giga GDVF103
- Silabs EFM32
- Broadcom bcm28/27 on RPi

Ref: linux driver for dwc https://github.com/torvalds/linux/tree/master/drivers/usb/dwc2

- implement and close #382 
- fix #1141 
- fix #126
- #750 should be re-test
closed 2021-11-04 by hathach