hathach/tinyusb #384: Allow overriding class drivers table

igrr, 2020-04-27:
At the moment, is is only possible to define one custom device class driver through custom_device.h/.c. When tinyusb is used as part of a larger application or framework, it becomes inconvenient since the specific driver to be used might only be known at run time, not at compile time.

It would be great if `static usbd_class_driver_t const usbd_class_drivers[]` would be made overridable by the application, without having to modify TinyUSB source code.

There are a few ways to achieve this, here is a relatively simple modification which should not affect the existing apps. It involves 3 changes:

1. Define each standard class driver structure as a separate const variable, expose this variable in the public header file.
2. Compose the default array of class drivers from the pointers to the class driver structures.
3. Allow the array of class drivers to be overridden.

Here is how it can look like:

```c++
  /* Define each class driver as a separate structure */
  #if CFG_TUD_CDC
    usbd_class_driver_t const tusb_cdc_class_driver = 
    {
        .class_code      = TUSB_CLASS_CDC,
        .init            = cdcd_init,
        .open            = cdcd_open,
        .control_request = cdcd_control_request,
        .control_request_complete = cdcd_control_request_complete,
        .xfer_cb         = cdcd_xfer_cb,
        .sof             = NULL,
        .reset           = cdcd_reset
    },
  #endif

  /* Same for the rest of the default class drivers */

  /* The default drivers table. Note that these are pointers to usbd_class_driver_t */
  static usbd_class_driver_t const* usbd_class_drivers_default[] = {
  #if CFG_TUD_CDC
    tusb_cdc_class_driver,
  #endif
  /* The rest of the default class drivers */
  };

  const usbd_class_driver_t **usbd_class_drivers = usbd_class_drivers_default;
  size_t usbd_class_drivers_count = sizeof(usbd_class_drivers_default)/sizeof(usbd_class_driver_t*);
```

In some public header file, expose the class driver structures, and the `usbd_class_drivers` variable:

```c++
    usbd_class_driver_t const tusb_cdc_class_driver;
    /* the rest of the class driver structures */

   const usbd_class_driver_t **usbd_class_drivers usbd_class_drivers;
   size_t usbd_class_drivers_count;
```

With this, the application can define a custom list of class drivers:

```c++
const usbd_class_driver_t** my_class_drivers = {
  tusb_cdc_class_driver,
  my_class_driver_1,
  my_class_driver_2
};

usbd_class_drivers = my_class_drivers;
usbd_class_drivers_count = sizeof(my_class_drivers) / sizeof(my_class_drivers[0]);
```

Alternatively, instead of exposing the array + size, it can be required to make the array terminated by a NULL pointer.

Instead of exposing global `usbd_class_drivers` and `usbd_class_drivers_count`, it is possible to expose a setter function.

hathach, 2020-04-28:
> At the moment, is is only possible to define one custom device class driver through custom_device.h/.c. When tinyusb is used as part of a larger application or framework, it becomes inconvenient [...]

This issue is raised before #313 by @pigrew, however, I still don't feel the need to do so, top of my head there is only 2 reasons:
- Implement a class driver that is not part of the stack
- Overwrite the current class driver in the stack

Both of this can be achieved by adding only 1 custom driver with latest code. Since usbd now allow driver's open() to claim interface (PR #342), custom driver can claim all interfaces it wants and divide it to appropriate implementation. Though I may be not fully understood your usage, could you give examples to elaborate it a bit more. I am open to any suggestions. 

> There are a few ways to achieve this, here is a relatively simple modification which should not affect the existing apps. It involves 3 changes:
>
> 1. Define each standard class driver structure as a [...]

Thank you for suggesting the implementation, it looks good, I will use this as reference if adding this is necessary.

igrr, 2020-04-28:
At the moment my use case is to add one non-standard custom driver (a debugging interface). I also expect that ESP-IDF users may want to implement other device class drivers which aren't supported in Tinyusb yet, for example UVC. Since we don't encourage modifications to the SDK source code or build system, we need to provide an extension point for users to add custom drivers. And in some cases, multiple custom drivers could be loaded (e.g. this debugging pipe, used by `idf.py`, and UVC).

> custom driver can claim all interfaces it wants and divide it to appropriate implementation. 

Thanks, i think this would work! Although the "custom" driver seems to be no longer present in master, what can we use instead of it?

One issue I see with the approach of using the custom driver as a dispatch point is that we still need to define some API for driver registration. Since the `usbd_class_driver_t` structure is private in `usbd.c`, i will have to define some other driver interface. As a result, drivers written by ESP-IDF users will not be directly pluggable into upstream TinyUSB, since the extension interface would be specific to ESP-IDF.

With that in mind, do you think it is worth adding an official extension point for registering custom drivers, or alternative implementations of the standard ones? Since this can be done in an almost zero-cost way just by changing how the driver table is defined, it seems like a good thing to do to avoid API fragmentation.

hathach, 2020-04-28:
> At the moment my use case is to add one non-standard custom driver (a debugging interface). I also expect that ESP-IDF users may want to implement other device class drivers which aren't supported [...]

Yeah, it makes sense. I would prefer UVC and most standard drivers is part of the repo. However, getting the stack and idf framework in sync constantly when adding new driver can be a PITA. Especially for driver that is targeted only esp32 and wouldn't care to run on other mcus/platforms. Although it is rather straight forward, I will probably need to write a API guidelines for it. Keep in mind that USBD-Class driver interface is considered internal so far, therefore it can be changed (but not a lot). 

> > custom driver can claim all interfaces it wants and divide it to appropriate implementation.
> 
> Thanks, i think this would work! Although the "custom" driver seems to be no longer present in master, what can we use instead of it?

Yeah, I removed the custom and replaced it with the vendor that use 0xFF, it has similar interface as CDC with buffered stream. You could actually use that for debug interface, though using custom maybe more preferred for you. 

> One issue I see with the approach of using the custom driver as a dispatch point is that we still need to define some API for driver registration. Since the `usbd_class_driver_t` structure is [...]

This makes sense as well, ok we can move `usbd_class_driver_t` to `usbd_pvt.h` (probably need to rename it to **usbd_class.h**) which can be included by application driver. I think we can have an additional API for register application driver array or maybe a weak array symbol (too static for app).

```
// specific name is TBD
tud_register_driver(usbd_class_driver_t[], count)
```
 
usbd will process with application drivers array first to allow overwrite the built-in, and they are kept separated from each other. register can be called anytime before the SET_CONFIG() preferably before enumeration started, else it needs to disconect/re-connect to take effect.

igrr, 2020-04-28:
The approach with tud_register_driver looks perfect!

Do you think it is okay to expose `usbd_class_driver_t` declarations of the standard drivers (CDC, MSC, ...) in usbd_pvt.h as well? It is not essential but would be nice when combining custom drivers with the drivers included in TinyUSB.

hathach, 2020-04-28:
> The approach with tud_register_driver looks perfect!
> 
> Do you think it is okay to expose `usbd_class_driver_t` declarations of the standard drivers (CDC, MSC, ...) in usbd_pvt.h as well? It is not essential but would be nice when combining custom drivers with the drivers included in TinyUSB.

It is OK, but I will need a reason to justify that, why would you want to call those driver function in application ? Those are when defined will always be const and won't be modifiable.

igrr, 2020-04-28:
Oh, sorry, i've missed this part:

> usbd will process with application drivers array first to allow overwrite the built-in, and they are kept separated from each other.

Since you intend to process  both the built-in drivers and the custom ones, we don't need to add the built-in drivers to the list passed to `tud_register_driver`. No problem then.

hathach, 2020-04-28:
Yeah, best to keep it separated, it it is rarely used feature as said. That allow the built-in to be able placed in ROM comfortably as it is :). I will make an PR for this shortly, though maybe @pigrew can have more to add to the discussion.

pigrew, 2020-04-28:
I don't have a clear plan right now. I'll think about it. Here are my initial thoughts.

Having both dynamic and static registration seems unnecessary to me. Is there a need for dynamic registration (instead of compile time)? I'm thinking no. 

However, I do like the idea of this being handled by the app layer. It already is to some extent (though the tusb_config header). The only disadvantage I see is that the compiler won't be able to optimize quite as well.

For ESP32, my thought is that the table is best defined in the app, so the USB library can be used as a static library with less component configuration. 

**Can the app pass the driver table as a parameter to usbd_init?**

Another question is if deinitialization needs to be supported. If so, we'd need to assert that the table's lifetime is longer than the lifetime of the tinyusb usage.

hathach, 2020-04-28:
> I don't have a clear plan right now. I'll think about it. Here are my initial thoughts.
> 
> Having both dynamic and static registration seems unnecessary to me. Is there a need for dynamic registration (instead of compile time)? I'm thinking no.

Even if the application doesn't have dynamic class driver, it doesn't make sense to me to have application to define all the class driver itself. The built-in will always be there within the stack. Application can just focus on class it specially want to implement. The implementation cost between static vs dynamic application driver is not that much.

> However, I do like the idea of this being handled by the app layer. It already is to some extent (though the tusb_config header). The only disadvantage I see is that the compiler won't be able to optimize quite as well.

gcc should be able to optimize when the class driver callback is not implemented, and remove all the code that loop over that array. 
 
> For ESP32, my thought is that the table is best defined in the app, so the USB library can be used as a static library with less component configuration.
>
> **Can the app pass the driver table as a [...]

much like the configuration and other descriptors, the stack will ask application for array of class driver when needed which happens every enumeration. There is no need for application to init or deinit that array.

The code probably speaks better than my English, I will do an PR.

me-no-dev, 2020-06-29:
Hey, I did a very simple modification that allows me to not load all included and enabled drivers and also load drivers of my own. The just of it is below. What do you think?

```cpp
#if CFG_TUSB_DYNAMIC_DRIVER_LOAD
static uint8_t USBD_CLASS_DRIVER_COUNT = 0;
static usbd_class_driver_t _usbd_driver[CFG_TUSB_DYNAMIC_DRIVER_MAX];

bool usbd_device_driver_load(const usbd_class_driver_t * driver)
{
    TU_VERIFY (USBD_CLASS_DRIVER_COUNT < CFG_TUSB_DYNAMIC_DRIVER_MAX);
    memcpy(&_usbd_driver[USBD_CLASS_DRIVER_COUNT++], driver, sizeof(usbd_class_driver_t));
    return true;
}
#else
static usbd_class_driver_t const _usbd_driver[] = {
#if CFG_TUD_CDC
    {
        DRIVER_NAME("CDC")
        .init             = cdcd_init,
        .reset            = cdcd_reset,
        .open             = cdcd_open,
        .control_request  = cdcd_control_request,
        .control_complete = cdcd_control_complete,
        .xfer_cb          = cdcd_xfer_cb,
        .sof              = NULL
    },
#endif
....
```

When the option is enabled, `usbd_class_driver_t ` structure and the new function are exposed to the user. The user code can then load the driver it needs before initialising TinyUSB.

hathach, 2020-06-29:
> Hey, I did a very simple modification that allows me to not load all included and enabled drivers and also load drivers of my own. The just of it is below. What do you think?

It is a good idea, however, it wouldn't help if user only want to overwrite e.g cdc and want to keep using the other built-in drivers such as msc. Which is why I made the PR #335

me-no-dev, 2020-06-29:
why not? you just do not load the included cdc driver and load your own cdc driver? what am I not getting?

me-no-dev, 2020-06-29:
in my case I need to have the drivers precompiled and load only what is necessary. Then load my own drivers as well. I do not want to overwrite any drivers in order to have a custom one.

hathach, 2020-06-29:
how do you use the built-in MSC driver and custom CDC driver in a composite device with above modification ?

hathach, 2020-06-29:
I see your point, but I don't like the approach, there are a few reasons
1. The built-in driver function e.g `vendord_init` is not mean to be used by application. I currently leave it in the header file for the sake of convenience, but could move it into usbd.c as private prototype when needed.
2. I prefer static over dynamic drivers
3. I prefer having _usbd_driver in ROM than SRAM since in general SRAM is much less than flash. It may not be an issue on your mcu/project, but it may be on other smaller mcu.

Overall, I will stick with the solution in the PR #335 which is simple enough and meet all the need.

me-no-dev, 2020-06-29:
sorry but it really does not meet my needs. do as you wish, but what I proposed is based on define switch and hurts nothing.
Given that you refuse, that means that I need to be looking again into custom copies of tinyusb files in IDF, in order to get it working properly for Arduino on ESP32S2. Your solution does not work at all in my case.

p.s. I have a couple of other PRs coming your way.

me-no-dev, 2020-06-29:
p.s.2 Your RAM/ROM concerns do not apply on ESP32S2.

hathach, 2020-06-29:
> sorry but it really does not meet my needs. do as you wish, but what I proposed is based on define switch and hurts nothing.
>
> Given that you refuse, that means that I need to be looking again into [...]

You are free to keep any modification in your own repo

tannewt, 2020-06-30:
> sorry but it really does not meet my needs. do as you wish, but what I proposed is based on define switch and hurts nothing.
> Given that you refuse, that means that I need to be looking again into [...]

Could you please comment on #385 with your use case that it wouldn't handle? Seems like it is close to done and just needs someone to add some last documentation.

me-no-dev, 2020-07-01:
@tannewt reading more into the PR is not that far from what I proposed. Main difference is that my proposal does not use callback to load the app drivers, does not load all built drivers and uses one list, as opposed to two. I will make work with what is made available and not stir more issues :)

hathach, 2020-07-01:
> @tannewt reading more into the PR is not that far from what I proposed. Main difference is that my proposal does not use callback to load the app drivers, does not load all built drivers and uses one list, as opposed to two. I will make work with what is made available and not stir more issues :)

I mentioned the PR multiple of times in our conversation :facepalm:

me-no-dev, 2020-07-01:
> I mentioned the PR multiple of times in our conversation 🤦

And I looked at it before you even mentioned it. It's not covering what I intended fully, but will make do. This is just saying "OK, I'll deal with it as is." facepalm was a bit unnecessary.

hathach, 2020-07-01:
Ok, fair enough. Make sure you put on comment what you like to add first in the PR. As mentioned above, I don't like dynamic driver loading at the moment.

me-no-dev, 2020-07-01:
I made tests with all and no drivers loaded. Difference was 300 or so bytes. Not worth it. Will go with all drivers loaded and then extra ones from the application through callback as you propose. I am looking forward being able to add custom drivers as well. Your PR will allow it. I am content :) nothing more to comment on it.

hathach, 2020-07-01:
> I made tests with all and no drivers loaded. Difference was 300 or so bytes. Not worth it. Will go with all drivers loaded and then extra ones from the application through callback as you propose. I [...]

Yeah, that is the reason I went with all drivers on smaller chip like nrf52 and samd. Sometimes we thought it is a good idea, but it could turn out not really. However, with more complicated driver such as Bluetooth HCI and/or RNDIS (which I think esp32s2 can make a good use of). It may worth a try. However, that is not worth our effort unless we could save at least 4-5KB of Flash. 

On the smaller MCU 300 bytes of SRAM is much more valuable on the ROM. TinyUSB aims to run on all the mcus therefore I am more careful on that. Thanks for the confirmation.

me-no-dev, 2020-07-01:
I will let you know once I look into BT and RNDIS, which I so definitely will. HCI makes no sense on S2, but it will on S3 :)
For RNDIS, i think that we have most things already covered in IDF (Lwip, etc) so I hope that will end up smaller in the end as well.

hathach, 2020-07-01:
> I will let you know once I look into BT and RNDIS, which I so definitely will. HCI makes no sense on S2, but it will on S3 :)
> For RNDIS, i think that we have most things already covered in IDF (Lwip, etc) so I hope that will end up smaller in the end as well.

There is no freertos port to run the rndis example, however, if you have any boards with stm32f4 (not sure if f103 could have enough ram/rom to run it ), you could try it out. You should be able to access the on-board webserver via USB with 192.168.7.1 

 https://github.com/hathach/tinyusb/tree/master/examples/device/net_lwip_webserver

me-no-dev, 2020-07-01:
I have 401 and 411RE, but both do not have USB i think. (Nucleo boards). One thing that comes to mind is to repurpose an old flight controller from one of my quads :) I will port it to S2 for sure. I think I will only have to mess with attaching to LwIP and that is it. It can be a native interface in IDF's IP stack, then be directly bridged to WiFi and more.

> stm32f4 has the same USB IP with your esp32s2 from synopsys, which I use a lot for developing the stack

I have the feeling that not everything is exactly the same for us, or at least does not fully work the same as documented. It could be hardware bugs also. If F4 uses the same IP, then that will be a nice device to test in parallel with. Case in point, that disconnect event...

hathach, 2020-07-01:
I already implemented that for your esp32s2, but it doesn't work out the same way as STM32F4

https://github.com/hathach/tinyusb/blob/master/src/portable/st/synopsys/dcd_synopsys.c#L749

That is possibly due to the lack of VBUS sense or something. I am not entirely sure, but the detach detection code work with stm32f4, you could also find this in reference code of stm32 cube library as well. For this, you really need to ping your hardware guy, he has the full access to Synopsys IP databook with detailed in register and how they work. Heck, you guys don't even release the USB document yet, what I have done have so far is totally based on STM USB documents.

me-no-dev, 2020-07-01:
I have the databook. It's what I used to figure out persistence. Thing is that I find differences from what it says there and what I see in reality. What it says there is probably in line with what you have on STM. IP Config here is a bit different, not all registers seem to respond to what is documented, and few more cases like that. I already poked the hardware guys for one edge case, will ask for the OTG interrupt. It seem to not come at all for ESP32S2. I was dumping all interrupt flags and suspend was all that came. To be fair, suspend is defined as no activity from the host for 3ms, so it is a valid ISR, just that is all that comes.

me-no-dev, 2020-07-01:
I wonder how possible it is to have an internal pullup activated for D+, causing the line to never go low when disconnected and that to not be triggering the proper interrupt... I will look into this.

me-no-dev, 2020-07-01:
yup... D+ is high after disconnect. This must be it :) will fix!

hathach, 2020-07-01:
> yup... D+ is high after disconnect. This must be it :) will fix!

I am glad you have access to the databook and figured it out, hardware is out of my territory. I leave this too you.

duempel, 2020-07-01:
> yup... D+ is high after disconnect. This must be it :) will fix!

I would wonder if ESP32-S2 OTG peripheral differs that much from STM. The _session end detected_ interrupt of STM is triggered due to voltage drop of Vbus. Also the D+ pull-up is just applied to the dataline if Vbus sensing detects a B-session valid voltage.

What's your exact hardware setup? Maybe the Vbus sensing pin sees a valid voltage even after disconnecting.

me-no-dev, 2020-07-01:
Let me see if we have vbus sense pin in the hardware, not that we can not repurpose one to drop the D+ pullup, but I was hoping that the peripheral will handle that on it's own (there is a separate register for USB io). By specs should the host hold D+ high or the device?

me-no-dev, 2020-07-01:
I see some OTG related IO signals defined for the matrix here: https://github.com/espressif/esp-idf/blob/5e436be77f68b1ba7d2ebbefb8aeda34f6c71276/components/soc/soc/esp32s2/include/soc/gpio_sig_map.h#L106-L124

Maybe `USB_OTG_VBUSVALID_IN_IDX ` will do the job

PR #385 (2020-08-11): support class drivers implemented by application
**Describe the PR**
fix #384 
Allow application to extend class driver that is not currently support by the stack. This is a draft PR, most of work is complete, though need feedback and a bit of polish for naming.

@pigrew @igrr
closed 2020-08-11 by hathach