hathach/tinyusb #394: Support using SEGGER_RTT_printf

xobs, 2020-05-08:
The tinyusb core now has partial support for SEGGER RTT.  However, support is incomplete, and it is not possible to use it with something such as circuit python.

This is caused by the definition of `tu_printf`:

```cpp
#ifndef tu_printf
    #define tu_printf           printf
#endif
```

The Segger RTT library has the following definition:

```cpp
int SEGGER_RTT_printf(unsigned BufferIndex, const char * sFormat, ...);
```

This is incompatible with printf due to the additional first argument.

It is possible to get this working by applying the following patch to `src/common/tusb_common.h`:

```patch
diff --git a/src/common/tusb_common.h b/src/common/tusb_common.h
index 052caef2..46468e5b 100644
--- a/src/common/tusb_common.h
+++ b/src/common/tusb_common.h
@@ -214,6 +214,8 @@ static inline bool     tu_bit_test (uint32_t value, uint8_t pos) { return (value
 #if CFG_TUSB_DEBUG

 void tu_print_mem(void const *buf, uint16_t count, uint8_t indent);
+#include "SEGGER_RTT.h"
+#define tu_printf(...) SEGGER_RTT_printf(0, __VA_ARGS__)

 #ifndef tu_printf
   #define tu_printf         printf
```

It would be nice if there was a CPP macro that did this.

hathach, 2020-05-08:
I will add support, but not as directly as this, you need to define the `tu_printf` though, since tinyusb source remain neutral and won't include any platform independent such as segger RTT header.

xobs, 2020-05-08:
Thanks.  Mostly the issue I ran into is that the prototype was different, so I couldn't just directly define it.  With the adapter macro, it worked flawlessly.

PR #395 (2020-05-08): add CFG_TUSB_DEBUG_PRINTF() for log retargeting
**Describe the PR**
Allow using custom printf function such as SEGERR_RTT_PRINTF() via macro define

**Additional context**
fix #394
closed 2020-05-08 by hathach