hathach/tinyusb #397: Restore IAR ARM support

HiFiPhile, 2020-05-08:
**Is your feature request related to a problem? Please describe.**
IAR ARM support was dropped in 9218d1e8bb9fd6cb37358c9e08f6ff9849c6110c. 

Currently IAR support GCC style function attributes, which makes the change much easier.

I've tested with IAR 8.50.1 and STM32L072KB with success.

**Describe the solution you'd like**
Restore IAR ARM support, I can make a pull request.

hathach, 2020-05-08:
yeah, I am definitely interested to get IAR support. Previously building with IAR is very annoying especially with pre packed & post packed things and I decide that is too much to support. I am glad they are going toward the standard gcc way, which is most opensource uses. I will be happy to review and merge the PR. If possible please include IAR as `CROSS_COMIPLE` since maintaining GUI IDE support cross platform is really a pain.

Btw do you, by any chance, know if IAR could correctly generate struct member alignment  as this 
(e.g CFG_TUSB_MEM_ALIGN is 64 byte aligned). I remembered got stuck by this with IAR on NXP LPC with this particular issue. 
https://github.com/hathach/tinyusb/blob/master/src/class/msc/msc_device.c#L50

hathach, 2020-05-08:
PS: is it possible to install IAR compiler on Linux. I just look at their website, there is an eclipse plugin, but I am not sure how it works. Not a problem if you don't know, I will need to do a research myself. I am hoping to get it install on linux ideally for ci build.

HiFiPhile, 2020-05-08:
> If possible please include IAR as CROSS_COMIPLE since maintaining GUI IDE support cross platform is really a pain.

First step I'll modify only the `tusb_compiler.h`, since it's enough for nornal use.

> PS: is it possible to install IAR compiler on Linux.

There is no native linux version, but I've installed it on Wine 5.0 with success, even debug works with J-Link tcp connection mode.

hathach, 2020-05-08:
thanks, that will help a lot adding more complete support later on.

hathach, 2020-05-08:
#398 should be enough to close this for now ? feel free to keep it open if you intend to submit more PRs.

closed 2020-05-08 by HiFiPhile