hathach/tinyusb #400: CDC DTR compatibility on Windows

HiFiPhile, 2020-05-09:
**Set up**
[Mandatory] Provide details of your setup help us to reproduce the issue as quick as possible  
 - **PC OS**   : Windows 10
 - **Board**   : STM32L072KB
 - **Firmware**: examples/device/cdc_dual_ports

**Describe the bug**
On Windows DTR bit is not set automatically when the port is opened, and some terminal programs don't set this bit. Which makes TinyUSB believe it's in unconnected state.

**To reproduce**
Steps to reproduce the behavior:
1. Launch HTerm 0.81.
2. Open the port and send some data.
3. See error.

To make it work, I used a ugly workaround, which ignores the DTR bit:
```
case CDC_REQUEST_SET_CONTROL_LINE_STATE:
    {
      // CDC PSTN v1.2 section 6.3.12
      // Bit 0: Indicates if DTE is present or not.
      //        This signal corresponds to V.24 signal 108/2 and RS-232 signal DTR (Data Terminal Ready)
      // Bit 1: Carrier control for half-duplex modems.
      //        This signal corresponds to V.24 signal 105 and RS-232 signal RTS (Request to Send)
      bool const dtr = tu_bit_test(request->wValue, 0);
      bool const rts = tu_bit_test(request->wValue, 1);

      p_cdc->line_state = (uint8_t) request->wValue | 0x01;
```

hathach, 2020-05-09:
Most terminal did, it is a well known norm. I know no alternatives to detect connection. I would suggest you to change terminal client or file issue to request the dtr implementation for that terminal

hathach, 2020-05-09:
Also please provide the log specifically when you connect and disconnect with your terminal

HiFiPhile, 2020-05-09:
> Most terminal did, it is a well known norm.

I agree with you, proper terminal clients handle this correctly. 

But there are some cases, especially some proprietary industrial control softwares doesn't handle it, they are hard to be replaced.

Silab, ST and J-Link's CDC implementation don't rely on DTR neither.

I suggest to add an option to bypass DTR detection.

> PS: when you connect/disconnect. Did the terminal set the rts bit ?

RTS is also not managed in my cases.

HiFiPhile, 2020-05-09:
> I suggest to add an option to bypass DTR detection.

In fact it's a feature request to make TinyUSB compatible with more terminal clients, rather than a bug.

hathach, 2020-05-09:
Unfortunately, It doesn't make much sense for me to add that option for now.

hathach, 2020-05-09:
Though I could review if you could make an PR for it with minimal changes to the code base.

HiFiPhile, 2020-05-09:
PR #401  submitted.

closed 2020-05-09 by HiFiPhile