hathach/tinyusb #403: Performance improvements in tusb_fifo

HiFiPhile, 2020-05-12:
**Is your feature request related to a problem? Please describe.**
Currently I'm making a USB-RS485 bridge, I noticed that at high baud rate(1M), data transmitted from RS485 is discontinued, seems like data receiving of  USB is slower than UART transmit, so I decided to do some performance investigation.

|Configuration ||
---|---
MUC|STM32L072KB
Frequency|32MHz
Compiler| IAR 8.50.1, High optimization

From the `cdc_dual_ports` example, I toggle a GPIO to show the USB receiving speed of 140 bytes:
```
while (1)
{
    tud_task();
    for (int itf = 0; itf < CFG_TUD_CDC; itf++)
    {
        if ( tud_cdc_n_available(itf) )
        {
            uint8_t buf[64];
            BSP_L_DE_GPIO->BSRR = 1 << BSP_L_DE_PIN;
            uint32_t count = tud_cdc_n_read(itf, buf, sizeof(buf));
            BSP_L_DE_GPIO->BRR = 1 << BSP_L_DE_PIN;
        }
    }
}
```
Between 2 packet receptions I got 650us, and I was a little surprised that `tud_cdc_n_read` execute for 240us !

Then I looked inside and found `tusb_fifo` is over-complicated in order to take care of different item sizes, but in general we fill only bytes, only exception is `osal_queue_t`.

**Describe the solution you'd like**

- Simply tsub_fifo to take care only bytes, and make changes in `osal`.

- Add functions to read/write bytes efficiently.

Here I tested the 2nd way, I created a function `tu_fifo_read_n_bytes`:
```
uint16_t tu_fifo_read_n_bytes (tu_fifo_t* f, void * buffer, uint16_t count)
{
  if( tu_fifo_empty(f) ) return 0;

  tu_fifo_lock(f);

  /* Limit up to fifo's count */
  if ( count > f->count ) count = f->count;

  if(count + f->rd_idx < f->depth )
  {
      memcpy(buffer, f->buffer + f->rd_idx, count);
      f->rd_idx += count;
  }
  else
  {
      memcpy(buffer, f->buffer + f->rd_idx, f->depth - f->rd_idx);
      memcpy(buffer, f->buffer, count - (f->depth - f->rd_idx));
      f->rd_idx = (f->rd_idx + count) % f->depth;
  }
  f->count -= count;
  
  tu_fifo_unlock(f);

  return count;
}
``` 
I replaced `tu_fifo_read_n` in `cdc_device.c`, now `tud_cdc_n_read` execute only for 25us, which is nearly 10x quicker. And interval between 2 packets went down to 515us.

hathach, 2020-05-13:
can you try to make _tu_ff_pull() and _tu_ff_push() as inline function, I am not convinced it is a good way to do. The default CFG_TUD_CDC_EPSIZE = 64 is not good for speed, since it effectively transit 1 packet per frame (1ms), try to increase it to 512 or something to transfer multiple packet per frame which is the actual speed-up 

There is issue for it #189 , though I haven't got time to test it out.

HiFiPhile, 2020-05-13:
> can you try to make _tu_ff_pull() and _tu_ff_push() as inline function

That's make nearly no difference. Now the problem is every time memcpy is called for transfer only one byte, for 64 bytes 64*memcpy are called. 
Also @kasjer made a good point for %(modulo), on Cortex-M0 it translate to `__aeabi_idiv` which cost ~50 cycles for every byte transferred.

### CFG_TUD_CDC_EPSIZE
Increase CFG_TUD_CDC_EPSIZE (and CFG_TUD_CDC_RX_BUFSIZE) improve greatly the performance.
For 700 bytes transferred, time decreased from 2ms to 626us.

[image]
[image]

hathach, 2020-05-13:
> That's make nearly no difference. Now the problem is every time memcpy is called for transfer only one byte, for 64 bytes 64*memcpy are called.

we can improve the current write_n()/read_n() by pre-computing the avaiable till the end of buffer (since it is circular fifo) ,and perform an single or at max 2 memcpy() for a single function call. It will be better than introducing new API(). I did have a plan to do that as well, just doesn't have enough time to do it.

> Also @kasjer made a good point for %(modulo), on Cortex-M0 it translate to `__aeabi_idiv` which cost ~50 cycles for every byte transferred.

Yeah, I literally doesn't give much attention to the speed much until now. We can replace modulo to `if larger then minus` which is probably be faster. But I am too lazy to set up something to test with. Please submit PR if you could, since you are got it all setup already.

> ### CFG_TUD_CDC_EPSIZE
> Increase CFG_TUD_CDC_EPSIZE (and CFG_TUD_CDC_RX_BUFSIZE) improve greatly the performance.
> For 700 bytes transferred, time decreased from 2ms to 626us.
> 
> 64 Bytes
> [image]
> 512 Bytes
> [image]

Great picture with annotation, it is a biggest bottleneck as suspected. There is a bit of modifying for using larger EP buffer. Let's me know if you encounter any issues with it.

HiFiPhile, 2020-05-13:
Beside these changes, I'd like also export FIFOs, 
```
tu_fifo_t* tud_cdc_get_rx_fifo(uint8_t itf)
{
    return &_cdcd_itf[itf].rx_ff;
}

tu_fifo_t* tud_cdc_get_tx_fifo(uint8_t itf)
{
    return &_cdcd_itf[itf].tx_ff;
}
```
In this way DMA (of UART for example) can access directly from/to the fifo and reduce a 2nd level buffer. 

Of cause we need to maintenant wr_idx & rd_idx.

hathach, 2020-05-13:
> Beside these changes, I'd like also export FIFOs,
>
> ```
> tu_fifo_t* tud_cdc_get_rx_fifo(uint8_t itf)
> {
>     return &_cdcd_itf[itf].rx_ff;
> }
>
> tu_fifo_t* tud_cdc_get_tx_fifo(uint8_t itf)
> {
>     return [...]

it won't be an option tu_fifo is only meant for internal usage only.

HiFiPhile, 2020-05-13:
> it won't be an option tu_fifo is only meant for internal usage only.

In this case we can also return a pointer of where wr_idx & rd_idx located in the FIFO. 

Less copy = more performance ;)

hathach, 2020-05-13:
the performance in question is too small to be considered,not mentioning mutex in preempted rtos, not all MCUs could do USB DMA to random unaligned SRAM. There is a risk that user overwrite other part of the fifo with memcpy on incorrect length. You can try to measure it, but I don't think we could get any significant number here.

HiFiPhile, 2020-05-13:
> not all MCUs could do USB DMA to random unaligned SRAM. 

USB DMA is not much concerned by this this change, it's used for generic DMA.
For example in a USB-UART bridge, UART DMA can read from the fifo.

It's totally fine that makes FIFO compatible with DMA, in our products we use them together intensively. 

> There is a risk that user overwrite other part of the fifo with memcpy on incorrect length.

Its also need to export maximum readable/writable linear length.

Anyway it's an advanced feature, anyone who use it should understand at least how FIFO and DMA works.

hathach, 2020-05-13:
such as driver doesn't need fifo in between, I think you better to implement your own cdc driver. Though it needs to wait for the #385 first. To me it doesn't make sense unless it get like triple the performance in overall system.

HiFiPhile, 2020-05-13:
> such as driver doesn't need fifo in between, I think you better to implement your own cdc driver.

Indeed that's make sense.

I just updated #404 if you can take a look.

HiFiPhile, 2020-05-15:
Optimizations have been made in #404 and #405.

closed 2020-05-15 by HiFiPhile