hathach/tinyusb #407: Scope all the includes in 'tinyusb' namespace

nitrotm, 2020-05-16:
When integrating this library into an existing project, it would be safer if all the includes were stored into a 'tinyusb' sub-folder. Also it would be nice to clearly separate headers into internal and external (implementation vs public api).

So instead of doing:
`
#include <tusb.h>
`

One could do:
`
#include <tinyusb/tusb.h>
`

nitrotm, 2020-05-22:
My use-case is maybe peculiar and I am not sure what's the vision behind tinyusb as there is direct support for specific boards and so potentially build embedded apps directly from this repo.
In my case, I'd like to embed TinyUSB the same way I integrated let's say lwIP or FreeRTOS. Basically compiling the necessary sources and making the APIs available to the user code through the includes.

As my framework grows, I realize there are lots of 3rdparty libraries which get included into the global namespace and potentially filename conflicts. What about `#include "common/sys_queue.h"`? In a large project, this could well be already there for another purpose.

lwIP implements such a strategy, where a user includes `#include "lwip/sys.h"` for instance.

But maybe all this points to the fact that a large project should compile external libs separately and wrap them into some kind of abstraction which doesn't expose the libs headers to user code (to avoid inevitable conflicts in a large project).

Currently, I moved the headers as described above and tweaked the code. Not ideal and will be a bit of a hassle to merge from upstream but I can live with it if nobody else has this issue.

hathach, 2020-05-23:
the bsp and example is needed for developing the stack. I have previously considered to have tinyusb core src to be on its own repo, however that requires lots more time maintaining the stack to keep everything in sync. You don't have to use any files other than one within the `src`. 

For includes, I like to have header next to its source rather than bundle everything in the same folder. Application only need to include "tusb.h", if there is an actual conflict within the stack with other, I will add prefix tusb_ to the file later on if needed.

closed 2020-05-23 by hathach