hathach/tinyusb #408: FreeRTOS tud_task() exit

nitrotm, 2020-05-16:
Currently `tud_task` is waiting on a queue indefinitely without any option to abort the loop. A solution would be to add an 'exit' message that can be sent to stop the loop.

Example:

```
void tud_stop()
{
  dcd_event_t event = { .rhport = rhport, .event_id = DCD_EVENT_EXIT, };

  dcd_event_handler(&event, false);
}
```

and handle the event in `tud_task`.

hathach, 2020-05-22:
currently, I don't see an reason to do this. 

PS: Next time please use the the feature template whenever you requested an new feature.

nitrotm, 2020-05-22:
@hathach I did choose 'feature' template but the label was not applied when I submitted. Maybe I should have kept the template text within?

My use-case is based on FreeRTOS and I spawn a dedicated task to handle usb core events in `tud_task`. But I also want to offer the option to turn off the usb stack (so being able to start or stop it dynamically). Therefore I implemented the `DCD_EVENT_EXIT` described above to signal the task waiting on the semaphore to wake-up and return.

hathach, 2020-05-23:
you could use the task handle to suspend and/or resume the usb task. currently TinyUSB doesn't support stop() yet since it requires lower level of dcd to turn off usb clock/ phy to save power. It is on my TODO, but not now.

closed 2020-05-23 by hathach