hathach/tinyusb #410: Implement get descriptor / device qualifier

nitrotm, 2020-05-16:
Most of the structure is there, it's only a matter of add a new callback and starting a transfer.

in `process_get_descriptor`:

```
    case TUSB_DESC_DEVICE_QUALIFIER:
    {
      TU_LOG2(" Device Qualifier\r\n");

      uint16_t len = sizeof(tusb_desc_device_qualifier_t);

      return tud_control_xfer(rhport, p_request, (void*) tud_descriptor_device_qualifier_cb(), len);
    }
```

hathach, 2020-05-23:
yeah, I am just being lazy since not many people need it. Will add it when having time.

hathach, 2020-05-28:
fixed by #421 but It is only step 1 of the process, after getting device qualifier successfully, host will ask for other speed configuration. Which is not supported currently, Other speed configuration will only be added when there is a testable example/setup

PR #421 (2020-05-28): add get device qualifier descriptor
**Describe the PR**
add `TU_ATTR_WEAK uint8_t const* tud_descriptor_device_qualifier_cb(void)` to support device qualifier. fix #410 

**Additional context**
It is only step 1 of the process, after getting device qualifier successfully, host will ask for other speed configuration. Which is not supported currently, Other speed configuration will only be added when there is a testable example/setup
closed 2020-05-28 by hathach