hathach/tinyusb #415: STM32 USB_OTG_FS without dedicated VBUS sense pin

UweBonnes, 2020-05-20:
Hello,

the stm32l476disco board has the dedicated VBUS sense pin connected as USART1 TX as default. VBUS sense is connected to GPIO PIN PC11 and hw/bsp/stm32l476disco/stm32l476disco.c configures PC11 as GPIO. However I do not see any later usage of PC11 and flashing the resulting executable resulting from "make BOARD=stm32l476disco DEBUG=1" makes the led blink. But I did not get the USB port to enumerates in many ways I tried, like reset, replug, repower the board etc.

Do I miss something?

hathach, 2020-05-20:
duplicated to #126, the board isn't working with tinyusb just yet. https://github.com/hathach/tinyusb/issues/126

closed 2020-05-20 by hathach

UweBonnes, 2020-05-20:
Drivers/STM32F7xx_HAL_Driver/Src/stm32f7xx_ll_usb.c  has following code for VBUS setup:
  /* VBUS Sensing setup */
  if (cfg.vbus_sensing_enable == 0U)
  {
    USBx_DEVICE->DCTL |= USB_OTG_DCTL_SDIS;

    /* Deactivate VBUS Sensing B */
    USBx->GCCFG &= ~USB_OTG_GCCFG_VBDEN;

    /* B-peripheral session valid override enable */
    USBx->GOTGCTL |= USB_OTG_GOTGCTL_BVALOEN;
    USBx->GOTGCTL |= USB_OTG_GOTGCTL_BVALOVAL;
  }
  else
  {
    /* Enable HW VBUS sensing */
    USBx->GCCFG |= USB_OTG_GCCFG_VBDEN;
  }

Blindly using this in the bsp file  makes the device enumerate:

  // Enable VBUS sense (B device) via pin PA9
//  USB_OTG_FS->GCCFG |= USB_OTG_GCCFG_VBDEN;
#define DEVICE_BASE     (USB_OTG_DeviceTypeDef *) (USB_OTG_FS_PERIPH_BASE + USB_OTG_DEVICE_BASE)
  USB_OTG_DeviceTypeDef * dev = DEVICE_BASE;
    dev->DCTL |= USB_OTG_DCTL_SDIS;

    /* Deactivate VBUS Sensing B */
    USB_OTG_FS->GCCFG &= ~USB_OTG_GCCFG_VBDEN;

    /* B-peripheral session valid override enable */
    USB_OTG_FS->GOTGCTL |= USB_OTG_GOTGCTL_BVALOEN;
    USB_OTG_FS->GOTGCTL |= USB_OTG_GOTGCTL_BVALOVAL;

Probably this is not the right place or procedure, but maybe people can help to get this right:

UweBonnes, 2020-05-20:
Obviuos a GPIO interrupt is missing to monitor VBUS. The setup does not reenumerate after a warm replug, bust starts with board reset.

hathach, 2020-05-20:
@UweBonnes let's move the dicussion to the other issue, there is an on-going discussion there already.