hathach/tinyusb #419: esp32s2 mdc example

chegewara, 2020-05-27:
Here is full info i could get from CDC/MSC device:
```
spressif Device
Manufacturer: Espressif Systems
Serial Number: 123456
Speed: 12Mb/s (full)
Bus:   1
Address:  10
USB Version:  2.00
Device Class: ef(misc )
Device Subclass: 02
Device Protocol: 01
Maximum Default Endpoint Size: 64
Number of Configurations: 1
Vendor Id: 303a
Product Id: 3000
Revision Number:  1.00

Config Number: 1
	Number of Interfaces: 3
	Attributes: a0
	MaxPower Needed: 100mA

	Interface Number: 0
		Name: cdc_acm
		Alternate Number: 0
		Class: 02(comm.) 
		Sub Class: 02
		Protocol: 00
		Number of Endpoints: 1

			Endpoint Address: 81
			Direction: in
			Attribute: 3
			Type: Int.
			Max Packet Size: 8
			Interval: 16ms

	Interface Number: 1
		Name: cdc_acm
		Alternate Number: 0
		Class: 0a(data ) 
		Sub Class: 00
		Protocol: 00
		Number of Endpoints: 2

			Endpoint Address: 02
			Direction: out
			Attribute: 2
			Type: Bulk
			Max Packet Size: 64
			Interval: 0ms

			Endpoint Address: 82
			Direction: in
			Attribute: 2
			Type: Bulk
			Max Packet Size: 64
			Interval: 0ms

	Interface Number: 2
		Name: usb-storage
		Alternate Number: 0
		Class: 08(stor.) 
		Sub Class: 06
		Protocol: 50
		Number of Endpoints: 2

			Endpoint Address: 03
			Direction: out
			Attribute: 2
			Type: Bulk
			Max Packet Size: 64
			Interval: 0ms

			Endpoint Address: 83
			Direction: in
			Attribute: 2
			Type: Bulk
			Max Packet Size: 64
			Interval: 0ms
```
It is detected properly by windows, but it takes 5-10 minutes before disk volume is detected and its not possible to access it.

chegewara, 2020-05-27:
Here is descriptor when i disable CDC:
```
spressif Device
Manufacturer: Espressif Systems
Serial Number: 123456
Speed: 12Mb/s (full)
Bus:   1
Address:  12
USB Version:  2.00
Device Class: 00(>ifc )
Device Subclass: 00
Device Protocol: 00
Maximum Default Endpoint Size: 64
Number of Configurations: 1
Vendor Id: 303a
Product Id: 3000
Revision Number:  1.00

Config Number: 1
	Number of Interfaces: 1
	Attributes: a0
	MaxPower Needed: 100mA

	Interface Number: 0
		Name: usb-storage
		Alternate Number: 0
		Class: 08(stor.) 
		Sub Class: 06
		Protocol: 50
		Number of Endpoints: 2

			Endpoint Address: 03
			Direction: out
			Attribute: 2
			Type: Bulk
			Max Packet Size: 64
			Interval: 0ms

			Endpoint Address: 83
			Direction: in
			Attribute: 2
			Type: Bulk
			Max Packet Size: 64
			Interval: 0ms
```

Its just for test, because i saw this difference when CDC is enabled:
`Device Class: ef(misc )`
and disabled:
`Device Class: 00(>ifc )`

chegewara, 2020-05-27:
Another question, not related. How we can change this value with standard, non vendor, descriptors:
`MaxPower Needed: 100mA`

hathach, 2020-05-27:
I don’t understand your “question” and what you are trying to tell here. Please try again with a bug template with all required field filled and a proper issue name

closed 2020-05-27 by hathach

chegewara, 2020-05-27:
I was trying to ask if this looks to you as good descriptor, but after some more digging and comparing with pendrive i answered myself. Thanks.

hathach, 2020-05-27:
For descriptor, please try to compile the “stock” in this repo which is `cdc_msc_freertos` and compare it, if you compiled from other repo that you are not sure about, you shouldn’t file an issue here.

chegewara, 2020-05-27:
Ive been trying to compile it, but im getting errors. So i give up and will try to figure out problem with esp32s2 and esp-idf.

hathach, 2020-05-27:
You could open an "bug" on that build error. People with similar build system could help you out