hathach/tinyusb #420: esp32 s2 saola dont mount volume in msc example

chegewara, 2020-05-28:
**Set up**
[Mandatory] Provide details of your setup help us to reproduce the issue as quick as possible  
 - **PC OS**   : Ubuntu 20.04 / Windows 10 / macOS 10.15 (all)
 - **Board**   : esp32s2 saola
 - **Firmware**: https://github.com/chegewara/esp-idf/tree/master/examples/peripherals/usb/tusb_sample_msc

**Describe the bug**
I dont know if its a bug. The problem i have is that msc volume does not mount on any system, even if driver is recognized properly (confirmed on windows and linux). On windows i can see in device manager drivers installed, then after some time (few minutes) i can see volume installed too for short time but i cant open it.

Here is wireshark capture (filter with `usb.addr == "1.75.0"`):
[msc-wireshark(1.75.0).pcapng.txt](https://github.com/hathach/tinyusb/files/4692882/msc-wireshark.1.75.0.pcapng.txt)

Full debug log from tinyUSB:
[debug_log.txt](https://github.com/hathach/tinyusb/files/4692885/debug_log.txt)

This log may contain logs from CDC too, which is not enabled in repository firmware i pointed, but CDC device works perfect.

hathach, 2020-05-28:
please try to see if you could reproduce the issue with this repo stock example, I don't have enough knowledge of your repo to troubleshoot. If you couldn't reproduce the issue with the stock example, the issue is probably within your example one.

hathach, 2020-05-28:
I just pulled out my saola board to test, it still run fine with the master using the stock example. TinyUSB issue is mostly for troubleshooting low level USB driver for MCU. Your issue sound like the problem with integrating TinyUSB into IDF, which you should file on ESP-IDF repo to get help from Espressif.

closed 2020-05-28 by hathach

chegewara, 2020-05-28:
Any help how to flash stock example?
When i am using it out of the box, then tinyUSB is disabled, even if example is compiled and flashed.
When i enable tinyUSB in menuconfig i get conflict with port files included in esp-idf.

hathach, 2020-05-28:
idf.py build
idf.py flash

hmm, after git pull eps-idf repo, the example doesn't work anymore. But it is a separated issue, I will create an issue for it.

chegewara, 2020-05-29:
Ok, i finally fixed it and it was not too hard. Will prepare esp-idf PR soon.