hathach/tinyusb #423: Teensy 4.0 board can't accept more than 64 bytes at a time on USB VCP

dpgeorge, 2020-05-28:
**Set up**
 - **PC OS**   : Arch Linux 
 - **Board**   : Teensy 4.0
 - **Firmware**: examples/device/cdc_msc

**Describe the bug**
With >64 bytes sent at once to USB VCP the board resets. 

**To reproduce**
1. Build `cdc_msc` for `teensy_40`
2. cat a file to the USB serial port, 64 byte file works fine, 65 byte file makes the board reset.

See also https://github.com/micropython/micropython/issues/6034

hathach, 2020-05-29:
thank you for filing the issue, I could reproduce it. The root cause is probably due to host sending more than EP size. Although we describe CDC data endpoint with packet size of 64, host driver possibly assume Highspeed device will use 512

Host send 69 bytes in one OUT token
[image]

Although it is allowable to declare packet size of 64 bytes in HS, and host must ensure not sending more than that . but I guess we have to increase as solution. Will do more testing for the fix.

From USB 2.0
[image]

hathach, 2020-05-29:
PS: Hmm, I think I misread the specs, sound like it must be 512 for Highspeed. Will do a update for other Bulk usage endpoint such as MIDI as well. 

```
The USB defines the allowable maximum bulk data payload sizes to be only 8, 16,
32, or 64 bytes for full-speed endpoints and 512 bytes for high-speed endpoints.
```

dpgeorge, 2020-06-01:
> it **MUST** be 512 for Highspeed.

Ahh, that would make sense.

alphaFred, 2020-06-02:
So this is just me hoping to learn something about tinyUSB. Hence please forgive my ignorance and maybe I am totally wrong here.

In some examples provided by tinyusb the following endpoint configuration can be found:
[image]
This looks to me as if the difference between FS and HS devices is take care of.

When I take a look at the source code in the MicroPython project (where this issue was brought to my attention) I can see the following:
[image]
[image]
So as of my understanding this should mean that the endpoint buffer is 256bytes. Which technically is not correct as of your excerpts from the USB standard but should allow for the 69bytes transfer. :thinking:

Is this configuration related to that issue or am I just showing my lack of knowledge about both tinyUSB and USB itself here :see_no_evil:

hathach, 2020-06-03:
> So this is just me hoping to learn something about tinyUSB. Hence please forgive my ignorance and maybe I am totally wrong here.
> 
> In some examples provided by tinyUSB the following configuration can be found:
> 
> ```c
> uint8_t const data_desc_configuration[] =
> {
>   // Interface count, string index, total length, attribute, power in mA
>   TUD_CONFIG_DESCRIPTOR(ITF_NUM_TOTAL, 0, CONFIG_TOTAL_LEN, TUSB_DESC_CONFIG_ATT_REMOTE_WAKEUP, 100),
> 
>   // Interface number, string index, EP Out & EP In address, EP size
>   TUD_MSC_DESCRIPTOR(ITF_NUM_MSC, 0, EDPT_MSC_OUT, EDPT_MSC_IN, (CFG_TUSB_RHPORT0_MODE & OPT_MODE_HIGH_SPEED) ? 512 : 64),
> };
> ```
> 
> This looks to me as if the difference between FS and HS devices is take care of.

Only for MSC, CDC should use the same method as well. However, the actual speed can be FS e.g in case teensy 4.0 is plugged into an FS only host. 

> When I take a look at the source code in the MicroPython project (where this issue was brought to my attention) I can see the following:
> 
> ```c
> #define CFG_TUD_CDC_RX_BUFSIZE  (256)
> #define CFG_TUD_CDC_TX_BUFSIZE  (256)
> 
> ...
> 
> static const uint8_t usbd_desc_cfg[USBD_DESC_LEN] = {
>     TUD_CONFIG_DESCRIPTOR(USBD_ITF_MAX, USBD_STR_0, USBD_DESC_LEN,
>         TUSB_DESC_CONFIG_ATT_REMOTE_WAKEUP, USBD_MAX_POWER_MA),
> 
>     TUD_CDC_DESCRIPTOR(USBD_ITF_CDC, USBD_STR_CDC, USBD_CDC_EP_CMD,
>         USBD_CDC_CMD_MAX_SIZE, USBD_CDC_EP_OUT, USBD_CDC_EP_IN, CFG_TUD_CDC_RX_BUFSIZE),
> };
> ```
> 
> So as of my understanding this should mean that the endpoint buffer is 256bytes. Which technically is not correct as of your excerpts from the USB standard but should allow for the 69bytes transfer. 
> Or is this just for the descriptor, essentially telling the host that the buffer is 256bytes large, but the actual buffer of tinyUSB is not of this size?

It is only the descriptor, the actual EP buffer has a different name EPSIZE, ( the RX/TX is the fifo size only).  https://github.com/hathach/tinyusb/blob/master/src/class/cdc/cdc_device.c#L64. You can try to increase this along with the RX/TX to see if that works

> Is this configuration related to that issue or am I just showing my lack of knowledge about both tinyUSB and USB itself here

Your observation is right, this is a bug in tinyusb. I will start to work on a fix next week or so with added API to get the actual speed (FS/HS) as well (currently occupied by other works).

alphaFred, 2020-06-07:
Hi @hathach I finally found the time to test your suggested fix.

I made the following changes in `tusb_config.h` of the micropython mimxrt port that I am working on.
From:
```C
#define CFG_TUD_CDC_RX_BUFSIZE  (256)
#define CFG_TUD_CDC_TX_BUFSIZE  (256)
```
To:
```C
#define CFG_TUD_CDC_RX_BUFSIZE  (512)
#define CFG_TUD_CDC_TX_BUFSIZE  (512)
#define CFG_TUD_CDC_EPSIZE (512)
```

Now it seems to work like a charm. All the tests I did previously with sending files larger then 64bytes work as expected. 
Additionally I tested with a significantly larger file, containing code for calculating a CRC with CRC table, having a total size of 2477 bytes. Hence this should result in multiple packets. Even that works perfectly.

Here's my crude code for testing ;-) 

```Python
"""
Test snippet for calulating table based 16bit CCIT CRC
"""

crc_table = (
0x0000, 0x1021, 0x2042, 0x3063, 0x4084, 0x50A5, 0x60C6, 0x70E7, 0x8108, 0x9129, 0xA14A, 0xB16B, 0xC18C, 0xD1AD, 0xE1CE, 0xF1EF,
0x1231, 0x0210, 0x3273, 0x2252, 0x52B5, 0x4294, 0x72F7, 0x62D6, 0x9339, 0x8318, 0xB37B, 0xA35A, 0xD3BD, 0xC39C, 0xF3FF, 0xE3DE,
0x2462, 0x3443, 0x0420, 0x1401, 0x64E6, 0x74C7, 0x44A4, 0x5485, 0xA56A, 0xB54B, 0x8528, 0x9509, 0xE5EE, 0xF5CF, 0xC5AC, 0xD58D,
0x3653, 0x2672, 0x1611, 0x0630, 0x76D7, 0x66F6, 0x5695, 0x46B4, 0xB75B, 0xA77A, 0x9719, 0x8738, 0xF7DF, 0xE7FE, 0xD79D, 0xC7BC,
0x48C4, 0x58E5, 0x6886, 0x78A7, 0x0840, 0x1861, 0x2802, 0x3823, 0xC9CC, 0xD9ED, 0xE98E, 0xF9AF, 0x8948, 0x9969, 0xA90A, 0xB92B,
0x5AF5, 0x4AD4, 0x7AB7, 0x6A96, 0x1A71, 0x0A50, 0x3A33, 0x2A12, 0xDBFD, 0xCBDC, 0xFBBF, 0xEB9E, 0x9B79, 0x8B58, 0xBB3B, 0xAB1A,
0x6CA6, 0x7C87, 0x4CE4, 0x5CC5, 0x2C22, 0x3C03, 0x0C60, 0x1C41, 0xEDAE, 0xFD8F, 0xCDEC, 0xDDCD, 0xAD2A, 0xBD0B, 0x8D68, 0x9D49,
0x7E97, 0x6EB6, 0x5ED5, 0x4EF4, 0x3E13, 0x2E32, 0x1E51, 0x0E70, 0xFF9F, 0xEFBE, 0xDFDD, 0xCFFC, 0xBF1B, 0xAF3A, 0x9F59, 0x8F78,
0x9188, 0x81A9, 0xB1CA, 0xA1EB, 0xD10C, 0xC12D, 0xF14E, 0xE16F, 0x1080, 0x00A1, 0x30C2, 0x20E3, 0x5004, 0x4025, 0x7046, 0x6067,
0x83B9, 0x9398, 0xA3FB, 0xB3DA, 0xC33D, 0xD31C, 0xE37F, 0xF35E, 0x02B1, 0x1290, 0x22F3, 0x32D2, 0x4235, 0x5214, 0x6277, 0x7256,
0xB5EA, 0xA5CB, 0x95A8, 0x8589, 0xF56E, 0xE54F, 0xD52C, 0xC50D, 0x34E2, 0x24C3, 0x14A0, 0x0481, 0x7466, 0x6447, 0x5424, 0x4405,
0xA7DB, 0xB7FA, 0x8799, 0x97B8, 0xE75F, 0xF77E, 0xC71D, 0xD73C, 0x26D3, 0x36F2, 0x0691, 0x16B0, 0x6657, 0x7676, 0x4615, 0x5634,
0xD94C, 0xC96D, 0xF90E, 0xE92F, 0x99C8, 0x89E9, 0xB98A, 0xA9AB, 0x5844, 0x4865, 0x7806, 0x6827, 0x18C0, 0x08E1, 0x3882, 0x28A3,
0xCB7D, 0xDB5C, 0xEB3F, 0xFB1E, 0x8BF9, 0x9BD8, 0xABBB, 0xBB9A, 0x4A75, 0x5A54, 0x6A37, 0x7A16, 0x0AF1, 0x1AD0, 0x2AB3, 0x3A92,
0xFD2E, 0xED0F, 0xDD6C, 0xCD4D, 0xBDAA, 0xAD8B, 0x9DE8, 0x8DC9, 0x7C26, 0x6C07, 0x5C64, 0x4C45, 0x3CA2, 0x2C83, 0x1CE0, 0x0CC1,
0xEF1F, 0xFF3E, 0xCF5D, 0xDF7C, 0xAF9B, 0xBFBA, 0x8FD9, 0x9FF8, 0x6E17, 0x7E36, 0x4E55, 0x5E74, 0x2E93, 0x3EB2, 0x0ED1, 0x1EF0
)

def calc_crc(data, start_val):
	crc = 0xFFFF & (start_val)
	for d in data:
		pos = 0xFF & (((0xFF) & (crc >> 8)) ^ ((0xFF) & d))
		print(pos)
		crc = (0xFFFF) & ((0xFFFF & (crc << 8)) ^ ((0xFFFF) & crc_table[pos]))
	return crc

test_data = [0x31, 0x32, 0x33, 0x34, 0x35, 0x36, 0x37, 0x38, 0x39]

print('{:02X}'.format(calc_crc(test_data, 0x0000)))
```

PR #464 (2020-07-19): Better support highspeed mode
**Describe the PR**
Fix #423, HS device can operate at FS mode e.g plugging into FS host. Therefore actual link speed should be checked when returning configuration descriptors and/or other related scenario when there is difference between fs/hs mode.
- add tud_speed_get()
- add dcd helper `void dcd_event_bus_reset (uint8_t rhport, tusb_speed_t speed, bool in_isr)`, which all highspeed core should report its speed after a bus reset.
- define both fs and hs configuration descriptor
- rename CFG_TUD_CDC_EPSIZE to CFG_TUD_CDC_EP_BUFSIZE with default size of 64 for FS [... cut]
closed 2020-07-19 by hathach