hathach/tinyusb #426: change CDC protocol on the fly

chegewara, 2020-06-01:
Can we add one more parameter to `TUD_CDC_DESCRIPTOR` macro to have control over comm protocol  from code?
https://github.com/hathach/tinyusb/blob/master/src/device/usbd.h#L197-L199

Thanks

pigrew, 2020-06-01:
@chegewara, what parameters are you wanting to change?

Also, you can fully define the descriptor in your application's code. There's no requirement to use the pre-defined macros.

chegewara, 2020-06-01:
This way:
```
#define TUD_CDC_DESCRIPTOR(_itfnum, _stridx, _ep_notif, _ep_notif_size, _epout, _epin, _epsize, protocol) \
  /* Interface Associate */\
  8, TUSB_DESC_INTERFACE_ASSOCIATION, _itfnum, 2, TUSB_CLASS_CDC, CDC_COMM_SUBCLASS_ABSTRACT_CONTROL_MODEL, protocol, 0,\
```

hathach, 2020-06-11:
why would you want to add the parameter ?

chegewara, 2020-06-11:
Because i dont know CDC protocol very good yet, and i see i can chose from that many options:
```
typedef enum
{
  CDC_COMM_PROTOCOL_NONE                   = 0x00 , ///< No specific protocol
  CDC_COMM_PROTOCOL_ATCOMMAND                     , ///< AT Commands: V.250 etc
  CDC_COMM_PROTOCOL_ATCOMMAND_PCCA_101            , ///< AT Commands defined by PCCA-101
  CDC_COMM_PROTOCOL_ATCOMMAND_PCCA_101_AND_ANNEXO , ///< AT Commands defined by PCCA-101 & Annex O
  CDC_COMM_PROTOCOL_ATCOMMAND_GSM_707             , ///< AT Commands defined by GSM 07.07
  CDC_COMM_PROTOCOL_ATCOMMAND_3GPP_27007          , ///< AT Commands defined by 3GPP 27.007
  CDC_COMM_PROTOCOL_ATCOMMAND_CDMA                , ///< AT Commands defined by TIA for CDMA
  CDC_COMM_PROTOCOL_ETHERNET_EMULATION_MODEL        ///< Ethernet Emulation Model
} cdc_comm_protocol_type_t;
```
I have prepared basic arduino library for esp32 and i would like to give users/programmers option to chose.

hathach, 2020-06-11:
that is not a good enough reason for me. The template is in user land, you can write your own to do that. I generally only added code when there is a good reason for it.

closed 2020-06-11 by hathach