hathach/tinyusb #428: Stopping and de-init

senilix, 2020-06-10:
Hi,

First, thanks for a great project. However, I have a question. I have a project where I would like to be able to stop and de-init the tinyusb stack and restart later on. 

I could of course just hack it and stop calling the tud_task function and de-initialize the hardware. But, it would be nice if one was able to do it without resorting to this. I run the tinyusb stack in a thread and my initial idea was that it would be nice if one could inject a stop-event into the tud_task queue requesting it to stop and perhaps the tud_task function could return a value that indicated the stack was stopped?

Any thoughts? 

Thanks

hathach, 2020-06-10:
This is thing I plan to do but haven't yet. Meanwhile you could try to call `tud_disconnect()` and/or adding your specific platform suspend task, turn off usb phy etc ... and do the opposite when calling `tud_connect()`

senilix, 2020-06-11:
Thanks, I will take a look. Can see I use a older version of tinyusb and will upgrade. But, it would be great if there were some structured way of closing down.

MarekRyn, 2024-08-05:
After calling `tud_disconnect()` it is not possible to restart connection with `tud_init()`. The reason is that global flag `_usbd_rhport` remains in valid state and function `tud_init()` returns on lines:
```
// skip if already initialized
if ( tud_inited() ) return true
```

The solution is to modify 'tud_disconnect()', as below:
```
bool tud_disconnect(void)
{
  TU_VERIFY(dcd_disconnect);
  dcd_disconnect(_usbd_rhport);
  _usbd_rhport = RHPORT_INVALID;
  return true;
}
```

JeffCalwood, 2025-02-28:
Isn't the counterpart to ```tud_disconnect()``` simply ```tud_connect()```? Adding ```_usbd_rhport = RHPORT_INVALID;``` to ```tud_disconnect()``` would break this pair of functions.
If you want to ```tud_init()``` after ```tud_disconnect()```, shouldn't you call ```tud_deinit()```?

roma-jam, 2025-02-28:
Hi, 
full `tusb_teardown()` implementation could be found in this PR:  https://github.com/hathach/tinyusb/pull/2904