hathach/tinyusb #429: IAD descriptor checks are too strict

PanRe, 2020-06-13:
**Set up**
[Mandatory] Provide details of your setup help us to reproduce the issue as quick as possible  
 - **PC OS**   : Windows 10
 - **Board**   : Custom STM32F411 board
 - **Firmware**: Custom Audio function

**Describe the bug**
I am currently working on an audio uac2 driver for tinyusb and therefore i need IAD descriptors. Now in the current implementation of usbd.c you enforce strict checks on the IAD descriptor as:

```c
// IAD's first interface number and class/subclass/protocol should match with opened interface
TU_ASSERT(desc_itf_assoc->bFirstInterface   == desc_itf->bInterfaceNumber   &&
                  desc_itf_assoc->bFunctionClass    == desc_itf->bInterfaceClass    &&
                  desc_itf_assoc->bFunctionSubClass == desc_itf->bInterfaceSubClass &&
                  desc_itf_assoc->bFunctionProtocol == desc_itf->bInterfaceProtocol);
```

Unfortunately those checks are too strict. As you mentioned in the comment, the above values "should" match but it is not mandatory. In the UAC2 definitions, the value for bFunctionSubClass of the IAD is defined to be AUDIO_FUNCTION_SUBCLASS_UNDEFINED (i.e. 0x00) and bInterfaceSubClass of the audio control (AC) interface (the immediatelly following interface descriptor) is AUDIO_SUBCLASS_CONTROL (i.e. 0x01). As such, the definitions in the UAC2 specifications provoke an error here. 

I would suggest to drop the subclass check! You may also find various examples (e.g. [here](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/usb-interface-association-descriptor)) where subclasses are not equal!

hathach, 2020-06-13:
You are right, we can drop the subclass and protocol check. Do you want to submit an PR for this ?

hathach, 2020-06-14:
close by #430

closed 2020-06-14 by hathach