hathach/tinyusb #436: midi tud_midi_rx_cb is not working

chegewara, 2020-06-18:
I have working midi code in arduino, and i connected esp32 s2 to web midi app. I know esp32 s2 is configured properly, because device is recognized by windows 10 and linux and i can play midi song from esp to PC on windows. With wrong descriptors drivers installation would fail.
Here is simple code that confirms `tud_midi_rx_cb` callback should be called but is not:
```
void tud_midi_rx_cb(uint8_t itf)
{
    Serial.println(__func__); // <--- this is not called
}

int MIDIusb::available()
{
    return tud_midi_available();
}
```

Here is log from available():
```
available: 16
available: 16
available: 16
available: 16
available: 16
```

hathach, 2020-06-18:
Please use the bug template those questions are mandatory

chegewara, 2020-06-18:
> Please use the bug template those questions are mandatory.

No need, i found the cause of issue, thanks.
Callback function is declared in header, but never invoked, so i added it here and now it works:
https://github.com/hathach/tinyusb/blob/master/src/class/midi/midi_device.c#L405

hathach, 2020-06-18:
> > Please use the bug template those questions are mandatory.
> 
> No need, i found the cause of issue, thanks.
> Callback function is declared in header, but never invoked, so i added it here and now it works:
> https://github.com/hathach/tinyusb/blob/master/src/class/midi/midi_device.c#L405

I am glad you figured it out, seem like it is on of the real bug, or maybe somehow removed in one of the PR. I will re-open it as reminder for fixing.

PR #497 (2020-08-27): Improve midi
**Describe the PR**
Enhance MIDI driver, help to address #495 
- fix #436 tud_midi_rx_cb() not invoked
- fix xfer_cb() not handle ep in
- add ZLP if needed

maybe help with #377  (worth a try)
commit a1b7e767 (2020-08-27): improve midi

- fix #436 tud_midi_rx_cb() not invoked
- fix xfer_cb() not handle ep in
- add ZLP if needed
closed 2020-08-27 by hathach