hathach/tinyusb #440: Clarify SAM D5x/E5x support?

jepler, 2020-06-23:
Over in CircuitPython we are adding support for the SAM E54.  Based on initial testing works properly with CFG_TUSB_MCU=OPT_MCU_SAMD51.  If no special support is required, perhaps the configuration variable could be renamed (we are saying SAM_D5X_E5X in CircuitPython) or it could be noted in the documentation.  However, it does not appear any change is _required_ at this time.

hathach, 2020-06-23:
Hi @jepler, I think it is best to have an additional option `OPT_MCU_SAME5X` Other vendors also use the same USB IP for different families. Normally it has a little different in configuration, included header, which is handled by dcd for example https://github.com/hathach/tinyusb/blob/master/src/portable/st/synopsys/dcd_synopsys.c#L35 .

In this case the asf4 does a great work of hardware abstraction, however, it is still beneficial to distinguish them for example, project that doesn't make use of asf4. I will make an PR to add the new MCU option.

hathach, 2020-06-25:
`OPT_MCU_SAME5X` and `OPT_MCU_SAMD51` are both existed although there is little difference thanks to asf4.

PR #441 (2020-06-25): added OPT_MCU_SAME5X option
**Describe the PR**
Try to fix #440 

@jepler for discussion, please let me know if this PR makes sense to you ? Let's me know if you have any other suggestion. 

PS: The `samd/dcd_samd.c`  should be renamed to `sam_de/dcd_sam_de.c` as well, though It will probably get renamed again when new series with same IP is added. So we could leave it as it is now, and will do a bit of google to see if we could figure out the microchip IP name (like dcd_synopsys of stm).
closed 2020-06-25 by hathach