hathach/tinyusb #448: Synopsys driver out endpoint handling broken by PR #399

kasjer, 2020-06-29:
**Set up**
 - **PC OS**   : Ubuntu 18.04 / Windows 10 (does not matter)
 - **Board**   : stm32f401blackpill (anything with synopsys)
 - **Firmware**: examples/device/cdc_msc

**Describe the bug**
#399 broke handling of OUT endpoints.
Code ignores data length in ***handle_epout_ints()*** and uses xfer->total_len even though packet was short,
later FIFO is filled with garbage based on this false total_len.
```c
          dcd_event_xfer_complete(0, n, xfer->total_len, XFER_RESULT_SUCCESS, true);
```

**To reproduce**
Steps to reproduce the behavior:
1. Build any CDC application
2. Hit enter to see `TinyUSB CDC MSC device example`
3. **Paste** something from clipboard or selection (on Linux)
4. What was copied should be visible fine
5. **Type** just one character, at this point previously copied data shows up again

hathach, 2020-06-30:
hmm, that is weird, I couldn't reproduce the issue with stm32f411disco board. Everything still work fine. I will try to find my 401 blackpill to test again.

hathach, 2020-06-30:
update: I didn't reproduce the exact behavior but found out another related bug that is pasting a long text will result in text got truncated in echo. E.g paste following text into the terminal (minicom on ubuntu 18.04) is fine pre PR #399 

```
// Stop transmitting packets and NAK IN xfers.
in_ep[epnum].DIEPCTL |= USB_OTG_DIEPCTL_SNAK;
while((in_ep[epnum].DIEPINT & USB_OTG_DIEPINT_INEPNE) == 0);

// Disable the endpoint.
in_ep[epnum].DIEPCTL |= (USB_OTG_DIEPCTL_STALL | USB_OTG_DIEPCTL_EPDIS);
while((in_ep[epnum].DIEPINT & USB_OTG_DIEPINT_EPDISD_Msk) == 0);
in_ep[epnum].DIEPINT = USB_OTG_DIEPINT_EPDISD;
```

After #399 the terminal can only echo

```
// Stop transmitting packets and NAK IN xfers.                                  
in_ep[epnum].DIEPCTL |= USB_OTG_DIEPCTDIEPINT_EPDISD;
```

I am doing more troubleshooting

kasjer, 2020-06-30:
I just checked STM32F407 target stm32f407disco, fresh master same behavior.
Tested with putty terminal, (minicom seems to send one byte at a time even when something is pasted into it).
Putty sends larger packets when something is pasted.

hathach, 2020-06-30:
> I just checked STM32F407 target stm32f407disco, fresh master same behavior.
> Tested with putty terminal, (minicom seems to send one byte at a time even when something is pasted into it).
> Putty sends larger packets when something is pasted.

Thank for the tips, I think it is probably a minor missed calculation of xfer len somewhere. I will try to fix it asap.

kasjer, 2020-06-30:
I think that now incoming packet size is ignored. xfre_len (which is size of endpoint size) is used to start transfer but transfer ends before when terminal is used and still xfer_len is used for transfer complete notification instead of short packet size.
Every packet however small is reported as 64 bytes.

kasjer, 2020-06-30:
I don't have access to USB analyzer but it would be visible I think that on each packet of size 1 (even if minicom is used) echo has 64 bytes, fist is true echo and the reset is just zeros that are probably ignored by minicom.

duempel, 2020-06-30:
Thanks for this information. I will try to reproduce this with my current setup.

`xfer->total_len` is the actual transfer length, not the max endpoint size. So also short length packages should still be handled fine. I'll have a look on it.

hathach, 2020-06-30:
> Thanks for this information. I will try to reproduce this with my current setup.
> 
> `xfer->total_len` is the actual transfer length, not the max endpoint size. So also short length packages should still be handled fine. I'll have a look on it.

Thanks @duempel I will leave this to you, ping me if you need any helps :smiley:

duempel, 2020-06-30:
Seems the problem is exactly as @kasjer described before. Since CDC class does not know how many bytes are going to be received it always prepares an out transaction of `TU_ARRAY_SIZE(p_cdc->epout_buf)` which is `CFG_TUD_CDC_EPSIZE`. The new DCD always invokes `dcd_event_xfer_complete` with the requested amount of bytes. It notices end of transfer due to short packets within PHY, but does not send this information back to stack.

```
  // Per USB spec, a short OUT packet (including length 0) is always
  // indicative of the end of a transfer (at least for ctl, bulk, int).
```
Since this old comment clearify that it's even said within spec, I'll write a fix for DCD driver. Sorry I missed this point at testing because I used PuTTY which does not print the NULL characters. HTerm shows all received bytes. Returning 1 character will result in one byte + 63 bytes of NULL (verified this on oscilloscope as well).

kasjer, 2020-06-30:
@duempel Funny thing is that I saw problem **because** I was using **PuTTY**, but I noticed it using Mynewt console first. Then I noticed that pasting (not typing) things results in some strange behavior. I hope fix will come faster then it took PR to go through :)

duempel, 2020-06-30:
> I hope fix will come faster then it took PR to go through :)

Luckily, I just had some time. Feel free to test #451 . 😉

closed 2020-06-30 by hathach