hathach/tinyusb #449: Type pipe_handle_t not found in HID class

mister-mojo, 2020-06-29:
**Set up**
[Mandatory] Provide details of your setup help us to reproduce the issue as quick as possible  
 - **PC OS**   : Ubuntu 18.04.4 LTS
 - **Board**   : Xilinx Zybo
 - **Firmware**: examples/host/cdc_msc_hid/

**Describe the bug**
Hey there, we are currently working on porting TinyUSB to the APU of the Xilinx Zybo to use a USB keyboard on it. The host driver seems to work okay so far. However, we are struggling with the HID class driver. When compiling the example code, there appears the following error:

In file included from /tinyusb/src/tusb.h:46:0,
                 from /tinyusb/hw/bsp/zybo/../board.h:42,
                 from /tinyusb/hw/bsp/zybo/zynq7000.c:29:
/tinyusb/src/class/hid/hid_host.h:199:3: error: unknown type name 'pipe_handle_t'
   pipe_handle_t pipe_hdl;

Seems like the type pipe_handle_t was never defined, since we could not find it anywhere. Do you have any hints about the intention of this type? 

**To reproduce**
To reproduce the behaviour, it should be sufficient to use any board with host capability and compile the firmware ``examples/host/cdc_msc_hid/`` with the parameter TUSB_HOST_HID set to 1 in the tusb_config. 

We'd really appreciate your help in fixing this issue. We're really looking forward to hearing from you!

hathach, 2020-06-30:
Yeah, it is unfortunately a legacy/obsolete code from host stack. In short `pipe_handle_t` is removed, and I already migrated the cdc and msc host driver away from that, but not HID. I was caught busy with other works and haven't got time to update the HID driver since then.

When I could manage my time, I will try to resume the work with host stack later on. Thanks for reporting the issue.

Klimm-Max, 2020-06-30:
Hi there,
i am facing the same issue with HID host driver.
it would be awesome to use the supported HID functionality. 👍

hathach, 2020-07-01:
I will try to look at the host stack asap to clean this up. Will let you know when it is ready, keep watching.

mister-mojo, 2020-07-03:
@hathach just started porting the HID class to the restructured host stack and compilation now passes successfully. Initialization seems to work, but there is no USB interrupt when pressing a key on a keyboard. Do you have any idea how to fix this? You may have a look on my code on the branch "zybo" in the forked repository at my account here on GitHub.

I'd really appreciate if you find some time in helping me fixing this.

mister-mojo, 2020-07-07:
Just introduced FreeRTOS into the project, since the Zynq-7000 platform is having problems when used in NO-OS mode. Now the keyboard is detected successfully all the tile, though some minor errors appear. Only the Interrupt mode seems to be not working yet. Looks like everything is registered fine, but no interrupts occur.

hathach, 2020-07-08:
@mister-mojo I would love to help, but I have no idea with the issue you are mentioning. If you could submit a PR for zynq7000 I could try it out to see what is wrong with the stack.

commit ef651e07 (2020-09-03): fix #449 remove obsolete pipehandle from hid host
PR #505 (2020-09-04): Update host
**Describe the PR**
Updated host stack as proof of concept, can mount msc + hid keyboard on both ehci and ohci.
Fix #449 

**Additional context**
Still lots of work with usbh and hcd
closed 2020-09-04 by hathach