hathach/tinyusb #450: DFU_RT Not receiving the callback

me-no-dev, 2020-06-29:
**Set up**
 - **PC OS**   :  macOS 10.15 
 - **Board**   : ESP32-S2-Saola
 - **Firmware**: examples/device/dfu_rt

**Describe the bug**
MAC OS seems to handle endpoint stall differently than other operating systems. The control requests that are received are as follows:
```
OUT Standard Interface 0xb 0 2 0
IN Class Interface 0x3 0 2 6
IN Class Interface 0x0 1000 2 0
```

If the first Control OUT is stalled, then the other requests will not come and the callback will not be called. TinyUSB's DFU_RT driver is looking only for the third packet, stalling the first two and resulting in fail.

**To reproduce**
Steps to reproduce the behavior:
1. Upload the example
2. Execute `dfu-util -U dump.dfu`
3. See error

**Example fix**
```cpp
bool dfu_rtd_control_request(uint8_t rhport, tusb_control_request_t const * request)
{
  // Mac Specific Handling
  bool is_mac_pre_packet = (request->bmRequestType_bit.type == TUSB_REQ_TYPE_STANDARD 
                         && request->bmRequestType_bit.recipient == TUSB_REQ_RCPT_INTERFACE 
                         && request->bRequest == 11);

  bool is_dfu_detach_cmd = (request->bmRequestType_bit.type == TUSB_REQ_TYPE_CLASS 
                         && request->bmRequestType_bit.recipient == TUSB_REQ_RCPT_INTERFACE 
                         && request->bRequest == DFU_REQUEST_DETACH);
  
  TU_VERIFY(is_mac_pre_packet || is_dfu_detach_cmd);
  
  tud_control_status(rhport, request);
  if(is_dfu_detach_cmd) {
      tud_dfu_rt_reboot_to_dfu();
  }
  return true;
}
```

hathach, 2020-06-30:
1. are running the actual `examples/device/dfu_rt`, that example doesn't support esp32s2 since it run on no-OS. If you made a modification to that example to run with esp32s2, please upload it here so that I could reproduce it as well.
2. please add the `#define CFG_TUSB_DEBUG   2` into the tusb_config.h and post your debug log here as well for analysis.

me-no-dev, 2020-06-30:
1. there is no meaningful modification done. Bug is discovered with vanilla TinyUSB master. The descriptor and code flow are exactly the same.
2. 

Unmodified log:
```
USBD Setup Received 80 06 00 03 00 00 FF 00 
  Get Descriptor String[0]
  Queue EP 80 with 4 bytes
USBD Xfer Complete on EP 80 with 4 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 80 06 00 03 00 00 FF 00 
  Get Descriptor String[0]
  Queue EP 80 with 4 bytes
USBD Xfer Complete on EP 80 with 4 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 01 0B 00 00 02 00 00 00 
  Set Interface
  dfu_rt control request
  Stall EP0
USBD Bus Reset
USBD Setup Received 00 05 0B 00 00 00 00 00 
  Set Address
USBD Xfer Complete on EP 80 with 0 bytes
USBD Setup Received 80 06 00 01 00 00 08 00 
  Get Descriptor Device
  Queue EP 80 with 8 bytes
USBD Xfer Complete on EP 80 with 8 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 80 06 00 01 00 00 12 00 
  Get Descriptor Device
  Queue EP 80 with 18 bytes
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 80 06 02 03 09 04 02 00 
  Get Descriptor String[2]
  Queue EP 80 with 2 bytes
USBD Xfer Complete on EP 80 with 2 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 80 06 02 03 09 04 16 00 
  Get Descriptor String[2]
  Queue EP 80 with 22 bytes
USBD Xfer Complete on EP 80 with 22 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 80 06 01 03 09 04 02 00 
  Get Descriptor String[1]
  Queue EP 80 with 2 bytes
USBD Xfer Complete on EP 80 with 2 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 80 06 01 03 09 04 24 00 
  Get Descriptor String[1]
  Queue EP 80 with 36 bytes
USBD Xfer Complete on EP 80 with 36 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 80 06 03 03 09 04 02 00 
  Get Descriptor String[3]
  Queue EP 80 with 2 bytes
USBD Xfer Complete on EP 80 with 2 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 80 06 03 03 09 04 24 00 
  Get Descriptor String[3]
  Queue EP 80 with 36 bytes
[... 161 lines cut ...]
Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Opening DFU capable USB device...
ID 303a:0002
Run-time device DFU version 0101
Claiming USB DFU Runtime Interface...
Determining device status: Device does not implement get_status, assuming appIDLE
Device really in Runtime Mode, send DFU detach request...
Resetting USB...
Opening DFU USB Device...
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 64
Copying data from DFU device to PC
Upload	[=========================] 100%      4194304 bytes
Upload done.
dfu-util: can't detach
Resetting USB to switch back to runtime mode
```

me-no-dev, 2020-06-30:
Modified log without actually rebooting:
```
USBD Setup Received 80 06 00 03 00 00 FF 00 
  Get Descriptor String[0]
  Queue EP 80 with 4 bytes
USBD Xfer Complete on EP 80 with 4 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 80 06 00 03 00 00 FF 00 
  Get Descriptor String[0]
  Queue EP 80 with 4 bytes
USBD Xfer Complete on EP 80 with 4 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 01 0B 00 00 02 00 00 00 
  Set Interface
  dfu_rt control request
  Queue EP 80 with zlp Status
USBD Xfer Complete on EP 80 with 0 bytes
USBD Setup Received A1 03 00 00 02 00 06 00 
  dfu_rt control request
  Stall EP0
USBD Setup Received 21 00 E8 03 02 00 00 00 
  dfu_rt control request
  Queue EP 80 with zlp Status
***DFU REBOOT***
USBD Xfer Complete on EP 80 with 0 bytes
USBD Bus Reset
...
```

hathach, 2020-06-30:
> 1. there is no meaningful modification done. Bug is discovered with vanilla TinyUSB master. The descriptor and code flow are exactly the same.

The log look good, however please pay attention to my comment, you are not running the `examples/device/dfu_rt` as claiming in the bug template, you are definitely running other application code judging from the log. 

Here is my request: **modify the stock dfu_rt example to run on the esp32s2**, you can also submit an pr to add `dfu_rt_freertos` or upload it here so that I could reproduce the issue. 

The stock example is preferred since it is simple enough to eliminate any other issues from other classes or complicated application. Hopefully this is clear to you.

me-no-dev, 2020-06-30:
Note: Enabling debug level 2 on ESP32-S2 and using `printf` will cause exceptions in interrupt debug prints. In order to collect the logs, I had to add the following change to tusb_common
```cpp
#ifdef CFG_TUSB_DEBUG_PRINTF
  extern int CFG_TUSB_DEBUG_PRINTF(const char *format, ...);
  #define tu_printf    CFG_TUSB_DEBUG_PRINTF
#else
  #include "esp32s2/rom/ets_sys.h" //changed
  #define tu_printf    ets_printf //changed
#endif
```

me-no-dev, 2020-06-30:
I am sorry. I wish I had the time to spend on creating new examples and providing you with a way to reproduce.
There is no way to currently add an example for esp32s2 without adding a very low level code to it, because vanilla TinyUSB have no way to init the USB pins and such code do not belong to examples. The only code that is called before `tusb_init()` is the code to setup the pins. Default descriptors are used for CDC and DFU (DFU should work with other drivers, correct?). That is it.

I have a limited number of other MCUs here that have usb, like Leonardo, Duo and BluePill STM32. Do you support out of the box any of those? Maybe I can confirm the situation on a different MCU as well?

hathach, 2020-06-30:
> I am sorry. I wish I had the time to spend on creating new examples and providing you with a way to reproduce.
> There is no way to currently add an example for esp32s2 without adding a very low level [...]

It is not that difficult you could build and flash with saola by go into https://github.com/hathach/tinyusb/tree/master/examples/device/cdc_msc_freertos and run. You obviously need to run idf export to have access to idf.py first. Though the dfu_rt example isn't added esp32s2 support like the cdc_msc_freertos yet. However, I totally understand if you don't have the time, I have little time myself.

```
make BOARD=esp32s2_saola_1 all flash
```

> I have a limited number of other MCUs here that have usb, like Leonardo, Duo and BluePill STM32. Do you support out of the box any of those? Maybe I can confirm the situation on a different MCU as well?

Yeah sure, any of supported MCUs will but more than OK to proof the point. Let's me know which board you are testing with and the usb stack log for the board as well. Here is the list of supported boards https://github.com/hathach/tinyusb/blob/master/docs/boards.md that I have access to.

me-no-dev, 2020-06-30:
How do you get the debug log from bluepill STM? I have the device running

me-no-dev, 2020-06-30:
To clarify, I tried all USB ports and seem to get no data.

me-no-dev, 2020-06-30:
Serial ports I mean... sorry.

me-no-dev, 2020-06-30:
Alright. STM is not my game, but I figured how to add serial output to the blue pill. Not sure how interested you are in the code.
Here is the log you requested. Case is the same:
```
USBD Setup Received 80 06 00 03 00 00 FF 00 
  Get Descriptor String[0]
  Queue EP 80 with 4 bytes
USBD Xfer Complete on EP 80 with 4 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 80 06 00 03 00 00 FF 00 
  Get Descriptor String[0]
  Queue EP 80 with 4 bytes
USBD Xfer Complete on EP 80 with 4 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 01 0B 00 00 00 00 00 00 
  Set Interface
  DFU-RT control request
  Stall EP0
USBD Bus Reset
...
```

hathach, 2020-06-30:
thank you for the detail information, I have put up another board to test with. The first 2 request is SET INTERFACE, and DFU GET STATUS. macOS seems to be more pickier than other. Although your fix get it work, those are std request within the specs and should be responded accordingly. I will pull out the DFU specs and put up a proper PR to fix this.

[image]

me-no-dev, 2020-06-30:
Agree :) 

What is that tool you are using to visualise the transactions?

hathach, 2020-06-30:
> Agree :)
> 
> What is that tool you are using to visualise the transactions?

It is lecroy usb protocol suite, you will need to have one of their USB protocol analyzer to capture the bus signal though

me-no-dev, 2020-06-30:
Wow! Their affordable solution is $1000.

hathach, 2020-07-01:
> Wow! Their affordable solution is $1000.

That is a very good price, it is a great portable equipment. can you try out the PR #452 to see if that fixes your issue.

me-no-dev, 2020-07-01:
> That is a very good price

I don't doubt it. Just I am not doing USB full time, so such investment makes no sence in may particular case :) I have a cheaper more general purpose LA that runs 9 channels at 200Msps and started on a decoder plugin. Will never be anything close to what your tool does, but could at least shed some light on issues on the bus for me.

>  can you try out the PR #452 to see if that fixes your issue.

Already did :) looks great, but seems unfinished (you added `dfu_status_t ` structure but did not use it in the code?)

hathach, 2020-07-01:
it is good enough for now, the dfu_status_t can be reused later with actual DFU mode.

PR #452 (2020-07-01): Fix dfu rt
**Describe the PR**
Fix #450
commit 12a145b2 (2020-07-01): fix dfu-rt to response to SET_INTERFACE and DFU_GETSTATUS

fix #450
closed 2020-07-01 by hathach