hathach/tinyusb #458: zephyr port

sslupsky, 2020-07-06:
Am am curious to know if you have considered porting TinyUSB to the Linux Foundation's embedded Zephyr RTOS?

hathach, 2020-07-08:
yeah, I would love to, but currently is too busy with other works and I am not familiar with zephyr as well. Maybe some time later this year.

josuah, 2023-10-02:
Did you know that this was done already?
[image]

https://www.youtube.com/watch?v=4Wv_X4J10Zk&t=1530

mabembedded, 2023-10-03:
Hello, I am chiming in here since I worked on this effort for a project and got a link to this issue from a colleague. I'll start the process to push my implementation of the port of TinyUSB to upstream Zephyr.

josuah, 2023-10-03:
One way to add a generic TinyUSB port would be the `device_next` driver API: it would give some generic `api->ep_enqueue()` method supported by all drivers, to use as an alternative to Zephyr own stack.
https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/usb/udc/udc_common.c#L568

Not sure about how Host mode works.

josuah, 2023-10-03:
On a side note: Kudos for your work and thank you for the talk! It helps for going down the same route, in particular for the debugging methods.

pauleffect90, 2023-10-10:
> Hello! I am chiming in here since I worked on this effort for a project and got a link to this issue from a colleague. I'll start the process to push my implementation of the port of TinyUSB to upstream Zephyr.

Thank you kindly for your work. You're awesome. Can't wait to finally be rid of that god-awful zephyr usb stack.

hathach, 2023-10-24:
superb !! Thank you very much @mabembedded for your effort :+1: . Eveen though I really want to, but unfortunately, I was a bit busy at the moment to follow up with Zephyr. I will try to catch up with you guys later.

RafalHalik, 2024-04-11:
> Hello! I am chiming in here since I worked on this effort for a project and got a link to this issue from a colleague. I'll start the process to push my implementation of the port of TinyUSB to upstream Zephyr.

Is this available anywhere?

OrhanYigitDurmaz, 2024-08-07:
> superb !! Thank you very much @mabembedded for your effort 👍 . Eveen though I really want to, but unfortunately, I was a bit busy at the moment to follow up with Zephyr. I will try to catch up with you guys later.

hey are you little free now? i would really like esp32s3 to have usb-otg support in zephyr

hathach, 2024-12-27:
zephyr port/integration will be my focus in 2025. Please let me know if anyone has any pending work/head-up that I could pick up. Thanks

tannewt, 2025-01-14:
I've started working on this as well for CircuitPython on Zephyr. I've got it working with a couple boards. My TinyUSB changes are pretty small: <https://github.com/hathach/tinyusb/compare/master...tannewt:tinyusb:zephyr> It is mostly the OSAL portions.

In CircuitPython I'm [initing the device](https://github.com/adafruit/circuitpython/compare/main...tannewt:circuitpython:zephyr#diff-41c9062a0ccc2825584273ce3de9f208f85921f8106abe3f1df1935486b271ef) and then [running a Zephyr thread to call tud_task](https://github.com/adafruit/circuitpython/compare/main...tannewt:circuitpython:zephyr#diff-68f04b426b838a98d4746f4c4d3178a931d816d270c453b6a0b7e48598043d33).

I think these would be good to incorporate into a Zephyr module. Instructions for that are here: https://docs.zephyrproject.org/latest/develop/modules.html

OrhanYigitDurmaz, 2025-01-14:
> I've started working on this as well for CircuitPython on Zephyr. I've got it working with a couple boards. My TinyUSB changes are pretty small: [...]

what soc's are supported right now?

tannewt, 2025-01-14:
> which soc's are supported right now?

I was testing on nRF52840 and STM32H575.

tannewt, 2025-01-17:
> Do you have an example project that i can use? i might try this with esp32s3

Not yet. Thach is working to get the TinyUSB samples working with Zephyr. My work is more hacky until that work is settled.

OrhanYigitDurmaz, 2025-01-17:
> > Do you have an example project that i can use? i might try this with esp32s3
> 
> Not yet. Thach is working to get the TinyUSB samples working with Zephyr. My work is more hacky until that work is settled.

is his changes on github on a branch or something?

tannewt, 2025-01-17:
> is his changes on github on a branch or something?

Not yet. He got pulled into a more urgent fix for the ESP32-S3.

OrhanYigitDurmaz, 2025-01-18:
> > is his changes on github on a branch or something?
> 
> Not yet. He got pulled into a more urgent fix for the ESP32-S3.

i think i saw that. DMA?

dhalbert, 2025-01-18:
The ESP32-S3 issue was #2493. It's mostly done -- probably requires a PR to ESP-IDF for a comprehensive fix.

MaheshAvula-Alifsemi, 2025-04-08:
Hi,
Has anyone tried porting the TinyUSB host stack to Zephyr, or can someone provide some pointers on how to get started with porting TinyUSB host stack support in Zephyr?"

OrhanYigitDurmaz, 2025-04-08:
> Hi,
> Has anyone tried porting the TinyUSB host stack to Zephyr, or can someone provide some pointers on how to get started with porting TinyUSB host stack support in Zephyr?"

There are some commits about zephyr support, but it's still not supported fully. Look at osal_zephyr.h file and corresponding portable files.

MaheshAvula-Alifsemi, 2025-04-08:
Hi @OrhanYigitDurmaz ,

Thank you for the responce, could you please  point me where the osal_zephyr.h available.