hathach/tinyusb #459: Why return false directly without leaving a callback to decide the result?

chang196700, 2020-07-06:
In file usbd.c:
Why return false directly without leaving a callback to decide the result?

Ref: https://github.com/hathach/tinyusb/blob/0c9932440b794fbdc699df6f0601f6febddedfe5/src/device/usbd.c#L881

hathach, 2020-07-07:
yeah, since I am too worried that application will use it as index which could overflow the memory since not everyone is aware of this microsoft extension. It should really be on its own callback, and I am too lazy to read and implement the microsoft specs example for that.

chang196700, 2020-07-08:
I will try to add a weak callback to provide the return value, but it may be a few days later.

hathach, 2020-07-08:
reopen since this is a legit feature. You could fix it with upcoming PR.

pigrew, 2020-07-09:
Perhaps just remove that whole if-branch, and always call tud_descriptor_string_cb? The application can handle the descriptor there, if it wants. The application should be doing bounds checking on the index, anyway. The TU_ASSERT would need to be a TU_VERIFY, so it wouldn't trigger a breakpoint if the string is out of bounds.

hathach, 2020-07-09:
@pigrew yeah, maybe I am over worried about application not being aware of this. But you are right, app should always do bound check anyway. And yes, with that change, the TU_ASSERT should be changed to TU_VERIFY.

PR #466 (2020-07-21): Remove 0xEE index string if statement & change TU_ASSERT to TU_VERIFY
**Describe the PR**
close #459
closed 2020-07-21 by hathach