hathach/tinyusb #460: Improve documentation for CDC  tud_cdc_write_available()?

CraigHutchinson, 2020-07-09:
**Is your feature request related to a problem? Please describe.**
When transmitting data over CDC the write operation can block if the transmit buffer is full and/or cannot fully contain the number of bytes to send. This can prevent preforming time critical functions etc.

**Describe the solution you'd like**
I would like to read the remaining capacity of the write buffer as a precondition to check. Only if there is capacity in the transmit buffer for a full message as to avoid blocking and postpone.

I found 'tud_cdc_write_available' but it is unclear if this function does as I need, maybe just needs rewording:
```
// Return number of characters available for writing
uint32_t tud_cdc_n_write_available (uint8_t itf);
```

Alternative to better documentation could be a more explicit function(s) for which there are options
```
// Check if a write to TX FIFO with bufsize bytes will be a blocking call
// - Blocking may occur when there is insufficient space in the transmit buffer for bufsize bytes
bool tud_cdc_will_n_write_block  (uint8_t itf, uint32_t bufsize);
```

```
// Write bytes to TX FIFO completely wihtout blocking
// @returns bufsize on success or 0 if not enough space for bufsize bytes in the FIFO to be non blocking
bool tud_cdc_n_write_noblock  (uint8_t itf, uint32_t bufsize);
```

Cheers.

duempel, 2020-07-09:
> I would like to read the remaining capacity of the write buffer as a precondition to check.

Hello Craig, you've already found the correct function for doing this. `tud_cdc_n_write_available()` actually returns the remaining capacity of items which fit into the dedicated fifo. Since CDC class defines the fifo's item size by 1, this function will return exactly the number of free bytes within the CDC tx buffer.
Do you have any suggestions for a more descriptive name of this function? Maybe _tud_cdc_n_write_free_buffer_size_ fits better than write_available 🤷‍♂️

Since there's already the described function I do not think there is any use introducing a new function to handle those special cases.

> write operation can block if the transmit buffer is full and/or cannot fully contain the number of bytes to send

What's your current setup? Are you using an OS or defined CFG_FIFO_MUTEX? The default setup of CDC will use a not overwriteable fifo. This means that if you put more bytes with `tud_cdc_n_write` than the remaining buffer capacity, the function will just truncate your data and let you loose some bytes (non blocking). If this function blocks the reason might be a locked fifo and not the actual amount of data to be transmitted.

CraigHutchinson, 2020-07-09:
@duempel Thanks for the response. I hadn't thought changing the function name itself but a improved comment could do it so as to not break existing code that may use it?

If I were to change the name somethign like this seems quite unambiguous:
`tud_cdc_get_tx_buffer_remaining`

> The default setup of CDC will use a not overwriteable fifo. This means that if you put more bytes with `tud_cdc_n_write` than the remaining buffer capacity, the function will just truncate your data [...]

Oh, interesting. I am getting blocked somehow else then. If the PC reading the data from the device goes a bit too slow and doesn't read the data my CDC device update or write appears to have been blocked. I had incorrectly assumed the write function would be blocking to put the data into the buffer.

However, with your clarification it actually could be the code calling the write that is retrying until all the data is written but I will have to check.

CraigHutchinson, 2020-07-09:
UPDATE: yes there is a loop counting until all the bytes are written so this is my blocking issue. However if I were to break at this condition I would have ended up with fragmented data so as per original-post would prefer not to not star writing if this fragmentation could occur.

CraigHutchinson, 2020-07-09:
@duempel I updated the description accordingly.

I think updating the comment in the code would help others and would remove ambiguity to this function being either:
1) The number of bytes available for transmitting onto the USB bus (i.e. How many bytes have been written to the buffer)
or
2) The number of bytes available to be written into the buffer (How many bytes can be written into the buffer before it is full)

hathach, 2020-07-09:
I wouldn't want to change the API name, it is similar to Arduino API, many Arduino user would find it familiar https://www.arduino.cc/reference/en/language/functions/communication/serial/availableforwrite/ 

> @duempel I updated the description accordingly.
>
> I think updating the comment in the code would help others and would remove ambiguity to this function being either:
>
> 1. The number of bytes [...]

1 is not technically correct, 2 is better with `The number of bytes available to be written into the FIFO`. Would you mind submitting the PR for this ?

PR #461 (2020-07-10): Improve comment on CDC tud_cdc_write_available()
Fixes #460

**Describe the PR**
A clear and concise description of what this PR solve.

**Additional context**
If applicable, add any other context about the PR and/or screenshots here.
commit 323ae5a8 (2020-07-10): Improve comment on CDC tud_cdc_write_available()

Fixes #460
closed 2020-07-10 by hathach