hathach/tinyusb #463: Auto-flush for cdc device

an-dr, 2020-07-13:
Hi! Is any chance to fix the flushing in some not-so-distant future?

https://github.com/hathach/tinyusb/blob/e1b3fe81c41d67ed033841a758c7e3b4d217a52b/src/class/cdc/cdc_device.c#L149-L155

It would be nice to have this logic in the driver. Thanks!

hathach, 2020-07-16:
the driver does auto-flush on complete, for now I don't really have an motive to implement and test this. Maybe later when I have more free time.

dobairoland, 2020-07-17:
@hathach My experience was that I had to flush manually after each write. Otherwise, only the first write was successful and the second one returned an error.

hathach, 2020-07-17:
> @hathach My experience was that I had to flush manually after each write. Otherwise, only the first write was successful and the second one returned an error.

can you provide a modified version of cdc_msc that causes the issue. The chance should be as minimal as possible.

dobairoland, 2020-07-17:
My original experiments were with ESP-IDF which uses an older version of tinyusb. Now I tried to reproduce the issue with the following code but I'm getting CPU panics all the time:
```c
void cdc_task(void* params)
{
  (void) params;

  uint8_t buf1[2] = {'a'};
  uint8_t buf2[2] = {'b'};

  // RTOS forever loop
  while ( 1 )
  {
    if ( tud_cdc_connected() )
    {
      // connected and there are data available
      if ( tud_cdc_available() )
      {
        int ret = tud_cdc_write(buf1, sizeof(buf1));
        /*tud_cdc_write_flush();*/
        ESP_LOGI("x", "ret = %d\n", ret);
        ret = tud_cdc_write(buf2, sizeof(buf2));
        /*tud_cdc_write_flush();*/
        ESP_LOGI("x", "ret = %d\n", ret);
      }
    }

    // For ESP32-S2 this delay is essential to allow idle how to run and reset wdt
    vTaskDelay(pdMS_TO_TICKS(10));
  }
}
```

8 times 2 bytes seem to be transferred and after that 0 bytes.
If I increase the buf1 and buf2 size to 64 then also only 16 bytes gets transferred and after that 0 bytes.

With flushing enabled used after each write: 2 bytes transferred, 2 bytes transferred, CPU panic; or 16 bytes transferred, 64 bytes transferred, CPU panic.

hathach, 2020-07-17:
@dobairoland If you believe the behavior is incorrect, please open your own issue using bug template and provide details there. This issue is simply a feature request. Note: you need to build with the master of this repo rather than version in the IDF.

PR #547 (2020-11-03): enable cdc auto flush on write()
**Describe the PR**
cdc write() auto call flush if there is enough data in the fifo. fix #463
closed 2020-11-03 by hathach