hathach/tinyusb #465: Compile issue with MSP430 gcc 9.2.0

hathach, 2020-07-19:
**Set up**
 - **PC OS**   : Ubuntu 20.04 
 - **Board**   : msp_exp430f5529lp
 - **Firmware**: examples/device/cdc_msc

**Describe the bug**
Failed to build due to a weird array-bound warning which I tried to fix but couldn't figure it out at this line
https://github.com/hathach/tinyusb/blob/master/src/portable/ti/msp430x5xx/dcd_msp430x5xx.c#L525

```
dcd_msp430x5xx.c:525:55: error: array subscript [65, 2097] is outside array bounds of 'volatile unsigned char[1]' [-Werror=array-bounds]
  525 |     volatile uint8_t * ep_buf = &USBSTABUFF + (ep_regs[BBAX] << 3);
      |                                                ~~~~~~~^~~~~~
```

gcc verion output

```
msp430-elf-gcc (Mitto Systems Limited - msp430-gcc 9.2.0.50) 9.2.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

**To reproduce**
1. Get the gcc 9.2.0 from TI https://www.ti.com/tool/download/MSP430-GCC-OPENSOURCE
2. Build example
3. See error

**Screenshots**
Not available

**Log**
Not available

cr1901, 2020-07-19:
@hathach I've made a compiler explorer project to experiment later. This is reduced down to the minimum: https://godbolt.org/z/WPsvbs

A workaround discussed on IRC (thanks @RichFelker) is:

```
#define EP_REGS(epnum, dir) (volatile unsigned char *)(uintptr_t)&USBOEPCNF_1 + 64*dir + 8*(epnum - 1)
```

hathach, 2020-07-21:
wow, thanks for the quick fix. I tried the walkaround on both compiler explorer and my real env for msp430, but it doesn't seems to fix the warning as well :D. Yeah, I also kind of guess `EP_REGS` cause the warning, but failed to see how to make gcc happy :)

duempel, 2020-07-23:
@cr1901 thanks for creating a compiler explorer project.

GCC 9 release seems to did some enhancements to existing warnings (quote from [GCC 9 Release](https://gcc.gnu.org/gcc-9/changes.html)):

> -Warray-bounds detects more instances of out-of-bounds indices.

But I have no idea what they actually changed. 😂

Maybe the warning/error is caused by volatile attribute. When changing this line
```
volatile uint8_t * ep_buf = &USBSTABUFF + (ep_regs[BBAX] << 3);
```
to
```
volatile uint8_t * ep_buf = &USBSTABUFF + (*(uint8_t *)(ep_regs + BBAX) << 3);
```
everything is fine for me. But this is not what you want. Doing so would cause access to the ep_regs+BBAX address to be handled as non-volatile. Maybe it's fine since this register is just set once at endpoint opening and the base address will never change. But I don't know the MSP430 driver well enough to estimate problems here. It's just to give you another idea. ;)

duempel, 2020-07-23:
I tried to figure out the cause of this problem and found [Bug 94675 - [9/10/11 regression] -Warray-bounds false positive with -O2 since r9-1948](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=94675). This seems to be related to our problem.

Maybe [this commit](https://gcc.gnu.org/git/gitweb.cgi?p=gcc.git;h=d893b683f40884cd00b5beb392566ecc7b67f721) can have brought up the false positives to gcc.

Also I did some very easy examples with compiler explorer that already threw the warning:
```
unsigned char c;
  
unsigned char f (void)
{
  unsigned char *p = &c;

  return *(p + 4);
}
```
The memory of object c seems to be checked. If we do not add the offset of 4 (or any other value) or declare c with a larger size, everything is working as expected. A new variable which stores the address or some typecasting as shown below can be used to loose track of the referenced object.
```
unsigned char c;
  
unsigned char f (void)
{
  unsigned char *p = &c;

  return *(unsigned char *)((long int)p + 4);
}
```
I don't know if this is handled as a bug and will be fixed in future releases or if it's a wished feature. Maybe I would just suppress the warnings right now and don't do changes to our sources?

hathach, 2020-07-24:
> I tried to figure out the cause of this problem and found [Bug 94675 - [9/10/11 regression] -Warray-bounds false positive with -O2 since r9-1948](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=94675). [...]

Great findings, very informative
 
> I don't know if this is handled as a bug and will be fixed in future releases or if it's a wished feature. Maybe I would just suppress the warnings right now and don't do changes to our sources?

-Warray-bounds is very useful warning, it may help to detect the actual out of bound array in the other parts of the stack. I prefer to have it enabled if possible 

> Edit: With these information I could change the example and get it compiled without any warnings (and even with output this time ):
> 
> ```
> volatile uint8_t * ep_buf = &USBSTABUFF + (*(volatile uint8_t *)((uintptr_t)ep_regs + BBAX) << 3);
> ```

Ah right, I was able to get pass the warning, it is actually very similar to @cr1901 walkaround. I see why it didn't work for my previously. Since I blindly replace the EP_REGS using USBOEPCNF_1, while the actual warning use USBSTABUFF. Thank you both very much, I will try to apply the fix asap. To be honest, gcc is outsmarted me on this.

> A workaround discussed on IRC (thanks @richfelker) is:
> 
> ```
> #define EP_REGS(epnum, dir) (volatile unsigned char *)(uintptr_t)&USBOEPCNF_1 + 64*dir + 8*(epnum - 1)
> ```

cr1901, 2020-07-24:
@duempel AFAIU, the warning in this case is a side-effect of the _real_ problem with my code as-written.

The original line is: `#define EP_REGS(epnum, dir) &USBOEPCNF_1 + 64*dir + 8*(epnum - 1)`

`USBOEPCNF_1` is declared as a `volatile unsigned char`. The compiler treats `USBOEPCNF_1` as a 1 byte object. When I take the address via `&USBOEPCNF_1`, I've created a `volatile unsigned char *`.  I am invoking UB by doing pointer arithmetic- `+ 64*dir + 8*(epnum - 1)`- to move past the storage allocated for `USBOEPCNF_1`, which is a single object _that has already been allocated by the compiler_.  This leads to the warning down the road; `gcc` kept track of the storage allocated originally for `USBOEPCNF_1` (1 byte), and complains when I try to do an out-of-bounds access on a pointer derived from `USBOEPCNF_1`. 

AIUI, casting to `(uintptr_t)&USBOEPCNF_1` makes the compiler "forget" about the storage allocated for the `USBOEPCNF_1` object. Since `uintptr_t` is guaranteed to be able to hold a pointer, we can do arithmetic on it, and cast to `volatile unsigned char *`. Our new `volatile unsigned char *` has been created without `gcc` thinking I was originally pointing to an object of only 1 byte. Thus I am free to use array indexing (equivalent to ptr arithmetic) to access anywhere within the I/O peripheral RAM starting at the address of `USBOEPCNF_1` :).

Short version: TI shouldn't have declared `USBOEPCNF_1` as `volatile unsigned char`. It should've been something like `#define USBOEPCNF_1 (*(volatile unsigned char *)0xdeadbeef)` (example address). However, I tried to use TI's definition and work around it. And it lead to code that worked in old compiler versions but was flagged as UB in newer versions :).

duempel, 2020-07-24:
@cr1901 thanks for this detailed description.

> Short version: TI shouldn't have declared USBOEPCNF_1 as volatile unsigned char. It should've been something like #define USBOEPCNF_1 (*(volatile unsigned char *)0xdeadbeef) (example address). [...]

Yes I agree with you. A different define could have been better at this point.

> casting to (uintptr_t)&USBOEPCNF_1 makes the compiler "forget" about the storage allocated for the USBOEPCNF_1 object

You made me very curious about this. I've been thinking about all the typecasting and how an object is referenced by gcc. I thought that the main problem was that the typecast was done within the `EP_REGS` define. Because we were creating a new `volatile uint8_t *` pointer with `ep_regs` I had the understanding that the pointed object is known to be of size `uint8_t` and that the read access to `ep_regs[BBAX]` was the main problem why I used pointer with typecasting here instead of array indexing. But this does not seem to make sense to me. So I tried out something:

```
#include <stdint.h>

typedef uint8_t * ep_regs_t;

uint8_t x;

uint8_t foo(void){
    ep_regs_t ep_regs = (uint8_t *)(uintptr_t)&x + 64;
    return ep_regs[1];      // We will fail here due to index != -64
}
```
This will bring up warnings as expected.
Changing line 8 to
```
ep_regs_t ep_regs = (uint8_t *)((uintptr_t)&x + 64);
```
will let us use `ep_regs` with any index we want!

Using
```
ep_regs_t ep_regs = (uint8_t *)((uintptr_t)&x);
```
only index 0 is working correctly.

This means that typecasting only is not making the compiler to forget about it. We have to add some arithmetic to it (otherway multiple castings would just optimized out I guess). So I could change @richfelker 's solution from instant multiple casting to
```
#define EP_REGS(epnum, dir) (ep_regs_t)((uintptr_t)&USBOEPCNF_1 + 64*dir + 8*(epnum - 1))
```

For me it's not throwing any warnings anymore. Can you approve this?

richfelker, 2020-07-24:
Yes, doing all the arithmetic as `uintptr_t` is preferable. Alternatively, if TI's headers could be fixed, the incorrect type `volatile unsigned char` could be changed to `volatile unsigned char []` (unspecified size) in which case the compiler would have to assume any non-negative offset may be valid.

Indeed my original solution is arguably invalid because the compiler can prove that the resulting pointer still points to the same object of size one, and any arithmetic *as pointers* outside of that is undefined. Arithmetic on `uintptr_t` is defined for all values, though.

hathach, 2020-07-27:
> Short version: TI shouldn't have declared `USBOEPCNF_1` as `volatile unsigned char`. It should've been something like `#define USBOEPCNF_1 (*(volatile unsigned char *)0xdeadbeef)` (example address). [...]

This is spot-on,  `USBOEPCNF_1` is linker symbol defined by PROVIDE, it should really be declared as pointer by TI. Following TI's definition is a natural way, and I think it is still best to stick with it and try to suppress the warning instead :) 

> This means that typecasting only is not making the compiler to forget about it. We have to add some arithmetic to it (otherway multiple castings would just optimized out I guess). So I could change [...]

Great findings, let's do it this way, gcc is very smart as always :) .  Thank you everyone for very helpful discussion :100:

PR #472 (2020-08-02): fix msp430 gcc 9.2.0 warning
**Describe the PR**
fix #465 warning with msp430 gcc to 9.2.0.050, also update ci cached toolchain as well
closed 2020-08-02 by hathach