hathach/tinyusb #473: a little help with assert fail

redmr2github, 2020-07-28:
**Describe what the question is**
Hi,

I didn't see forum or something of a wiki, so asking maybe a simple question regarding getting tinyusb integrated into a stm32 project.  If there is a better medium, please let me know.

I have been evaluating USB on stm32h747 device on the stm32h747eval board - specifically cdc and msc.  The ST driver/stack with freely available arm eabi gcc toolchain seems to perform poorly.  This has been confirmed by ST.  So, I'm trying to see if tinyusb stack may do better.  I started with using example code - cdc-msc meant for stm32h743 using directions from https://github.com/hathach/tinyusb/blob/master/docs/porting.md - with minimal edit to use only the M7 core on stm32h747.  That worked for awhile.  To say, I can reproduce all the problems another user reported with cdc demo, and I can read the readme.txt mounted as mass storage device.  But this morning, it stopped working.  The debugger shows infinite loop trying to execute HAL_PWREx_ConfigSupply.  There is really no explanation, considering it used to work just find before last power cycle.  I thought maybe I fried the board, but loading other images seems to be ok.

That rerouted my effort to just trying to load tinyusb to a existing simple project as explained here - https://github.com/hathach/tinyusb/blob/master/docs/getting_started.md.  After a few bumps on the road with clocks and interrupts, it finally executes tusb_init.  I'm not using OS, so I have tud_task in a while (1) loop.  The debugger stops at assert failures from tud_task, and I'm not sure how to proceed from here.  I get the feeling it might be something simple to someone who's familiar with the code.  I'm just trying to stay afloat.  Following is the output when I use LOG=2, and attached are the tusb_config.h and main.cpp files
[tusb_config.h.txt](https://github.com/hathach/tinyusb/files/4991381/tusb_config.h.txt)
[main.cpp.txt](https://github.com/hathach/tinyusb/files/4991389/main.cpp.txt)

.  The usb_descriptor.c and msc_dks.c from cdc-msc demo are copied over are unchanged.  Can anyone please help with what may be going on here?  Thanks.

USBD init
CDC init
MSC init
USBD Bus Reset 

USBD Setup Received 00 05 17 00 00 00 00 00 
  Set Address
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 80 06 00 01 00 00 12 00 
  Get Descriptor Device
  Queue EP 80 with 18 bytes
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 02 03 09 04 02 00 
  Get Descriptor String[2]
  Queue EP 80 with 2 bytes
USBD Xfer Complete on EP 80 with 2 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 02 03 09 04 1E 00 
  Get Descriptor String[2]
  Queue EP 80 with 30 bytes
USBD Xfer Complete on EP 80 with 30 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 01 03 09 04 02 00 
  Get Descriptor String[1]
  Queue EP 80 with 2 bytes
USBD Xfer Complete on EP 80 with 2 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 01 03 09 04 10 00 
  Get Descriptor String[1]
  Queue EP 80 with 16 bytes
USBD Xfer Complete on EP 80 with 16 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 03 03 09 04 02 00 
  Get Descriptor String[3]
  Queue EP 80 with 2 bytes
USBD Xfer Complete on EP 80 with 2 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 03 03 09 04 0E 00 
  Get Descriptor String[3]
  Queue EP 80 with 14 bytes
USBD Xfer Complete on EP 80 with 14 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 02 00 00 09 00 
  Get Descriptor Configuration[0]
  Queue EP 80 with 9 bytes
USBD Xfer Complete on EP 80 with 9 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 02 00 00 62 00 
  Get Descriptor Configuration[0]
  Queue EP 80 with 64 bytes
USBD Xfer Complete on EP 80 with 64 bytes
  Queue EP 80 with 34 bytes
USBD Xfer Complete on EP 80 with 34 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 00 09 01 00 00 00 00 00 
  Set Configuration
  Open EP 81 with Size = 8
  Open EP 02 with Size = 512
  Open EP 82 with Size = 512
  Queue EP 02 with 512 bytes … OK
  CDC opened
process_set_config 755: assert failed
process_control_request 582: assert failed
  Stall EP0

USBD Setup Received 00 09 01 00 00 00 00 00 
  Set Configuration
cdcd_open 245: assert failed
process_set_config 798: assert failed
process_control_request 582: assert failed
  Stall EP0

USBD Setup Received 00 09 01 00 00 00 00 00 
  Set Configuration
cdcd_open 245: assert failed
process_set_config 798: assert failed
process_control_request 582: assert failed
  Stall EP0

hathach, 2020-07-29:
This looks more like bug report than a question, please update the post to use bug template, and also change the title to better reflect your problem.

redmr2github, 2020-07-29:
This is going to sound stupid, but while I was cleaning up the project files and build scripts to make a tarball for bug submission - it now works!  The only change in the source files was in the main.cpp file where I moved a printf statement...  I'll monitor it for a few days to make sure things are stable.

But to answer your questions -
I don't have a supported board handy - but with minor edits to stm32h743eval files, and I can get the stock cdc_msc example code to work on stm32h747eval.  The infinite loop executing HAL_PWREx_ConfigSupply that I had reported seems to go away if booting from flash.

Thanks!

hathach, 2020-07-30:
I am glad you work it out, that is why I always ask user to reproduce the issue with stock example + minimal changes on the supported board. In general, I can only help with usb issue, any customization with mcu clock, hw modification, flash config etc ... is out of knowledge. My advice whenever you have an issue, just start again from stock example, and try to do bisect from that.

Closed since this issue is resolved.

closed 2020-07-30 by hathach