hathach/tinyusb #481: STM32L0 WebUSB example: no data is received

aauer1, 2020-08-05:
**Set up**
I compiled the webusb_serial example for the STM32L053 discovery board.
 - **PC OS**   : Ubuntu 18.04.4
 - **Board**   : STM32L053 discovery
 - **Firmware**: examples/device/webusb_serial

**Describe the bug**
CDC is working as expected (ttyACM0 device is available and data is sent back on the CDC device and is also visible on the webusb example website)
But if I enter something on the webpage which should be transmitted to the board, it is not sent back. It seems that no data is received at all.

**To reproduce**
Steps to reproduce the behavior:
1. Go to the webusb example website
2. Connect the TinyUSB device
3. Enter something on the website and hit enter
4. It's expected to receive the entered data on the website again (echo) but nothing appears

**Log**
I don't see any output on the log, when I enter some data on the website.

hathach, 2020-08-05:
You could enable the stack debug log with `LOG=2`  It provides internal states of the stack for analyzing.  https://github.com/hathach/tinyusb/tree/master/examples#log

I don't have L053 to test with, but I will try to help you with  the progress. Would you mind submitting the PR to add the board, that will help other with actual hardware to verify/test with.

aauer1, 2020-08-05:
I used the stm32l0538disco board which is already part of your source tree. 

This is the debug output when the controller starts and is attached to the USB port of my PC. I think this is the normal enumeration process:
<pre>
USBD init
CDC init
VENDOR init
USBD Bus Reset
USBD Bus Reset
USBD Setup Received 80 06 00 01 00 00 40 00 
Request Type: 80  Get Descriptor Device
  Queue EP 80 with 18 bytes
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Bus Reset
USBD Setup Received 00 05 1A 00 00 00 00 00 
Request Type: 0  Set Address
USBD Xfer Complete on EP 80 with 0 bytes
USBD Setup Received 80 06 00 01 00 00 12 00 
Request Type: 80  Get Descriptor Device
  Queue EP 80 with 18 bytes
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 80 06 00 06 00 00 0A 00 
Request Type: 80  Get Descriptor Device Qualifier
  Stall EP0
USBD Setup Received 80 06 00 06 00 00 0A 00 
Request Type: 80  Get Descriptor Device Qualifier
  Stall EP0
USBD Setup Received 80 06 00 06 00 00 0A 00 
Request Type: 80  Get Descriptor Device Qualifier
  Stall EP0
USBD Setup Received 80 06 00 02 00 00 09 00 
Request Type: 80  Get Descriptor Configuration[0]
  Queue EP 80 with 9 bytes
USBD Xfer Complete on EP 80 with 9 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 80 06 00 02 00 00 62 00 
Request Type: 80  Get Descriptor Configuration[0]
  Queue EP 80 with 64 bytes
USBD Xfer Complete on EP 80 with 64 bytes
  Queue EP 80 with 34 bytes
USBD Xfer Complete on EP 80 with 34 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 80 06 00 03 00 00 FF 00 
Request Type: 80  Get Descriptor String[0]
  Queue EP 80 with 4 bytes
USBD Xfer Complete on EP 80 with 4 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 80 06 02 03 09 04 FF 00 
Request Type: 80  Get Descriptor String[2]
  Queue EP 80 with 30 bytes
USBD Xfer Complete on EP 80 with 30 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 80 06 01 03 09 04 FF 00 
Request Type: 80  Get Descriptor String[1]
  Queue EP 80 with 16 bytes
USBD Xfer Complete on EP 80 with 16 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 80 06 03 03 09 04 FF 00 
Request Type: 80  Get Descriptor String[3]
  Queue EP 80 with 14 bytes
USBD Xfer Complete on EP 80 with 14 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 00 09 01 00 00 00 00 00 
Request Type: 0  Set Configuration
  Open EP 81 with Size = 8
  Open EP 02 with Size = 64
  Open EP 82 with Size = 64
  Queue EP 02 with 64 bytes ... OK
  CDC opened
  Open EP 03 with Size = 64
  Open EP 83 with Size = 64
  Queue EP 03 with 64 bytes ... OK
  VENDOR opened
  Queue EP 80 with zlp Status
USBD Xfer Complete on EP 80 with 0 bytes
USBD Setup Received 80 06 04 03 09 04 FF 00 
Request Type: 80  Get Descriptor String[4]
  Queue EP 80 with 24 bytes
USBD Xfer Complete on EP 80 with 24 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
USBD Setup Received 21 20 00 00 00 00 07 00 
Request Type: 21  CDC control request
  Set Line Coding
  Queue EP 00 with 7 bytes
USBD Xfer Complete on EP 00 with 7 bytes
  CDC control complete
  Queue EP 80 with zlp Status
USBD Xfer Complete on EP 80 with 0 bytes
USBD Setup Received 80 06 05 03 09 04 FF 00 
Request Type: 80  Get Descriptor String[5]
  Queue EP 80 with 30 bytes
USBD Xfer Complete on EP 80 with 30 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
</pre>

This is the debug output when I connect on the website with webusb:
<pre>
USBD Setup Received 01 0B 00 00 02 00 00 00 
Request Type: 1  Set Interface
  VENDOR control request
  Stall EP0
USBD Setup Received 02 01 00 00 03 00 00 00 
Request Type: 2  Clear Feature
  Queue EP 80 with zlp Status
  VENDOR control request
  Queue EP 80 with zlp Status
USBD Xfer Complete on EP 80 with 0 bytes
USBD Setup Received 02 01 00 00 83 00 00 00 
Request Type: 2  Clear Feature
  Queue EP 80 with zlp Status
  VENDOR control request
  Queue EP 80 with zlp Status
USBD Xfer Complete on EP 80 with 0 bytes
USBD Setup Received 21 22 01 00 02 00 00 00 
Request Type: 21  VENDOR control request
  Queue EP 83 with 33 bytes ... OK
  Queue EP 80 with zlp Status
USBD Xfer Complete on EP 80 with 0 bytes
USBD Xfer Complete on EP 83 with 33 bytes
  VENDOR xfer callback
</pre>

But if I enter some data on the website nothing happens on the debug output. No new line is printed.

hathach, 2020-08-05:
I don't see anything abnormal in the log. Did the webusb landing page appear, and you could connect/pair with device. The webusb is still also experimental in chrome. Could you pull out another PC preferably on different OS to test with.

aauer1, 2020-08-05:
Landing page appears and it is also possible to connect/pair with the device. And also the vendor control requests are working fine.  But there seems to be no callback/event/... for received data.
I will try it on another PC.

hathach, 2020-08-05:
it is probably udev, permission, and set-up things. It should probably work on windows.

aauer1, 2020-08-05:
I just tried it on Windows and it really works there. So, Windows and Linux are somehow handling this WebUSB stuff differently.

hathach, 2020-08-05:
it is not a surprise, it is probably permission things ( check your udev and chrome docs etc ..) The feature is still experimental and may or may not work on some setup.

aauer1, 2020-08-05:
I'm not sure if this is a OS thing. Because I already have WebUSB running together with an STM32L401 controller and your stack.

hathach, 2020-08-05:
do your udev include the permission for VID/PID of the example ? If not, please try to add it, it may be needed for chrome to claim the interface.

xmos-jmccarthy, 2021-03-09:
I ran into the exact same issue while porting for another microcontroller.  The demo worked on Windows.  On Linux it would enumerate but not be recognized in Chrome.

Assuming you kept the default descriptors, adding this rule to udev may resolve your Linux issues as well:
ATTRS{idVendor}=="cafe", ATTRS{idProduct}=="4011", MODE="0666"

closed 2021-03-10 by hathach

hathach, 2021-03-10:
thanks @xmos-jmccarthy for confirming the root cause. I update the docs regarding vendor/webusb vs udev to help others in the future.

PR #710 (2021-03-10): update doc for BOARD name, add note for udev on Linux
**Describe the PR**
Update docs to mention how to find out `BOARD` and udev for webusb example. Fix  #481