hathach/tinyusb #488: the MSC sd card RW speed is too slow,at about 200K/S

Thanos-Love-Ironman, 2020-08-18:
Hi
MCU:stm32h743
Demo:CDC_MSC_FreeRTOS
I rewrite the msc_disk.c with ST HAL sd card driver.But it's RW speed is too slow,at about 200K/S
Is it the TinyUSB issue or ST HAL sd card driver issue?
what RW speed that the tinyusb msc suppose to be?
Thank you.

ladyada, 2020-08-18:
you should test ST HAL SD card driver and then report back the max speed for writing you have there.

coledd, 2020-08-18:
You can improve the USB transfer speed by adjusting the endpoint buffer size in tusb_config.h. It is set very low by default to fit most microcontrollers. But for the H7, you can bump this value up higher to improve throughput. Adjust to your liking, for example:

// MSC Buffer size of Device Mass storage
#define CFG_TUD_MSC_EP_BUFSIZE   4096

Thanos-Love-Ironman, 2020-08-20:
> you should test ST HAL SD card driver and then report back the max speed for writing you have there.

Thank you. 
The TinyUSB is working well.But the ST HAL SDcard driver cannot working when using dma type function.I am still working on it

chegewara, 2020-08-20:
Hi,
i am not following tests procedures here, but using my library for esp32 S2 on arduino i could achieve this performance:
1) this example is using RAM disk and internal flash:
[image]

2) this example is using SD card over SPI:
[image]

Thanos-Love-Ironman, 2020-08-20:
> Hi,
> i am not following tests procedures here, but using my library for esp32 S2 on arduino i could achieve this performance:
> 
> 1. this example is using RAM disk and internal flash:
>    [image]
> 2. this example is using SD card over SPI:
>    [image]

That's really fast!
My cdc_msc_freertos project using ST HAL Library sd card driver just can't reach that speed.
In fact,it event cannot transfer data fluently,the RW speed is up to 350KB/S,down to 0KB/S sometimes.

chegewara, 2020-08-20:
I still think that esp32 s2 can do better. I have to connect LA and see real SPI speed.

hathach, 2020-08-20:
closed since this is not tinyusb issue but ST HAL driver one, and I don't have any knowledge with ST Hal to help with. If you want to benchmarch the stack performance, you should do it with msc ramdisk by reporting large enough size, and faking msc read/write API.

closed 2020-08-20 by hathach