hathach/tinyusb #496: STM32F2/STM32F4 workaround for USB334x HS fail

diggit, 2020-08-24:
**Describe the bug**
It is already known in STM32 community, that F2 and F4 MCUs fail to work with USB334x (USB3341, USB3343, USB3346, USB3347) (and maybe others) external PHYs in HS mode and fall back to FS. It is timing problem on ULPI, more in [module 2 in USB334x errata](http://ww1.microchip.com/downloads/en/DeviceDoc/80000645A.pdf). Workaround was introduced by enabling bit `USB_OTG_DCFG_XCVRDLY` as mentioned in [this community forum thread](https://community.st.com/s/question/0D50X00009XkZUeSAN/stm32f2f4-problems-with-various-ulpi-usb-phys) (you have to load more replies, button on page bottom).
Affected devices:
- stm32f401
- stm32f405
- stm32f407
- stm32f411
- stm32f412
- stm32f413
- stm32f415
- stm32f417
- stm32f423
- stm32f427
- stm32f429
- stm32f437
- stm32f439
- stm32f446
- stm32f469
- stm32f479
- stm32f205
- stm32f207
- stm32f215
- stm32f217

So basically whole F4 and F2 line featuring USB HS.

**Solution**
As mentioned, solution is to enable `XCVRDLY` in `DCFG` register. Tested and working with STM32F446 and USB3343 PHY by setting this bit in `set_speed` function.

I don't have patch/PR, because I am not familiar with coding style in tinyusb and have limited time to study it.

hathach, 2020-08-24:
thank you for opening the issue with useful and helpful reference link. Currently I don't have any of above hardware combination to start working on a PR. I will try to find one later on.

diggit, 2021-02-20:
Is there any way, I can help with this issue?

hathach, 2021-02-22:
@diggit yeah, if you are willing to make a PR for this, I will be very happy to merge the it.

zhm-nico-nico-ni, 2021-02-23:
F469 F749   has the "Bit 14 XCVRDLY", but others not have in OTG_HS_DCFG.
Can you share other solution?

diggit, 2021-02-24:
F446 definitely has this bit. It was quite recent addition at that time. Try updating ref. manuals, headers and SVD files. ULPI has much more configs, but ST does not document it. I do not guarantee that it's there, but AFAIK this issue is present on all F2, F4 and maybe F7 MCUs.
I am not aware of any other solution.

tin-, 2021-11-11:
Solution with enabling XCVRDLY in DCFG register worked great with STM32H743ZIT and USB3370B PHY by setting this bit in set_speed function. Thanks!

hathach, 2021-11-12:
@tin- I have no issue with H743 tested on stm32h743 evaluation board (as well as other openh743 + phy add-ons). This issue probably does not affect the H743

diggit, 2021-11-12:
This issue happens with USB33xx phys and synopsys usb core.

hathach, 2021-11-13:
@diggit I have no problem getting stm32h743eval running with highspeed with on-board USB3320 PHY. It appear only has issue with more recent PHY. I will test to see if setting XCVRDLY still work with currently-working 3320 PHY. If not, then we can safely always set it to 1.

[image]

hathach, 2021-11-15:
#1195 is up to fix this, please try it out since I don't have this combination to test with. The fix is applied to the new synopys/dwc2 driver (old stm/synopsys will be fading out in the future).

closed 2021-11-19 by hathach

diggit, 2021-11-19:
I've tested #1195 and it solves this issue for my current HW setup (STM32H730 + USB3343)

hathach, 2021-11-19:
thank you very much for your confirmation.

PR #1195 (2021-11-19): set DCFG_XCVRDLY when using external ULPI highspeed phy
**Describe the PR**
Fix #496 by setting DCFG_XCVRDLY when using with external ULPI phy. It has no impact with H743 and currently-working older phy USB3320 (probably a fraction of ms delay). So I guess it is safe to just set it in all cases.