hathach/tinyusb #498: "NetworkError: Unable to set device interface." when testing webusb_serial on stm32f407disco

crosswick, 2020-08-26:
**Set up**
[Mandatory] Provide details of your setup help us to reproduce the issue as quick as possible  
 - **PC OS**  : macOS 10.14.6 
 - **Board**   : stm32f407disco
 - **Firmware**: examples/device/webusb_serial

**Describe the bug**
I've compiled the webusb_serial example as described in the readme and uploaded it with STM32CubeProgrammer. Then, with the board powered via the same USB connection (with an AC charger brick) and the mini-USB port connected to my laptop, the Chrome popup does come up: "Go to www.tinyusb.org to connect."

Once there however, next to the Connect button it says "NetworkError: Unable to set device interface.". When clicking on the Connect button, that error goes away. When I then type someting in the Sender box and hit enter, nothing happens (the text should come back in the Receiver box, right?). When I hit Connect again, select 'TinyUSB Device - Paired' and hit Connect there, again the same error appears.

I have tried with several different PORT= settings to no avail. I don't yet understand how to construct a log file. I hope this is a simple and known issue... thank you.

hathach, 2020-08-27:
You could enable the stack debug log with LOG=2 It provides internal states of the stack for analyzing. https://github.com/hathach/tinyusb/tree/master/examples#log

may related to #481

crosswick, 2020-08-27:
Alright I've got the logging working; was a bit of a discovery for me to realize that I needed an extra serial-to-USB converter, and to find out which pins on the board I needed, and that the datarate should be 115200.

Here is the output; at the end is where I tried to connect by clicking the Connect button in the browser.

By the way: I have another piece of tinyusb/webusb hardware here that does immediately connect and work (https://webusb.hackz.one/)

USBDUSBD init
CDC init
VENDOR init
USBD Bus Reset 

USBD Setup Received 00 05 18 00 00 00 00 00 
  Set Address
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 80 06 00 01 00 00 08 00 
  Get Descriptor Device
  Queue EP 80 with 8 bytes
USBD Xfer Complete on EP 80 with 8 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 00 01 00 00 12 00 
  Get Descriptor Device
  Queue EP 80 with 18 bytes
USBD Xfer Complete on EP 80 with 18 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 02 03 09 04 02 00 
  Get Descriptor String[2]
  Queue EP 80 with 2 bytes
USBD Xfer Complete on EP 80 with 2 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 02 03 09 04 1E 00 
  Get Descriptor String[2]
  Queue EP 80 with 30 bytes
USBD Xfer Complete on EP 80 with 30 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 01 03 09 04 02 00 
  Get Descriptor String[1]
  Queue EP 80 with 2 bytes
USBD Xfer Complete on EP 80 with 2 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 01 03 09 04 10 00 
  Get Descriptor String[1]
  Queue EP 80 with 16 bytes
USBD Xfer Complete on EP 80 with 16 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 03 03 09 04 02 00 
  Get Descriptor String[3]
  Queue EP 80 with 2 bytes
USBD Xfer Complete on EP 80 with 2 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 03 03 09 04 0E 00 
  Get Descriptor String[3]
  Queue EP 80 with 14 bytes
USBD Xfer Complete on EP 80 with 14 bytes
  Queue EP 00 with zlp Status
USBD Xfer Complete on EP 00 with 0 bytes
[... 246 lines cut ...]
  Queue EP 80 with zlp Status
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 21 20 00 00 00 00 07 00 
  CDC control request
  Set Line Coding
  Queue EP 00 with 7 bytes
USBD Xfer Complete on EP 00 with 7 bytes
  CDC control complete
  Queue EP 80 with zlp Status
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 21 20 00 00 00 00 07 00 
  CDC control request
  Set Line Coding
  Queue EP 00 with 7 bytes
USBD Xfer Complete on EP 00 with 7 bytes
  CDC control complete
  Queue EP 80 with zlp Status
USBD Xfer Complete on EP 80 with 0 bytes

USBD Setup Received 01 0B 00 00 02 00 00 00 
  Set Interface
  VENDOR control request
  Stall EP0

crosswick, 2020-08-27:
[image]
this is how my board shows up in System Information

hathach, 2020-08-28:
> USBD Setup Received 01 0B 00 00 02 00 00 00
Set Interface
VENDOR control request
Stall EP0

Seem like it stall the SET_INTERFACE, I don't have a mac to test with. I will try to revise the issue and give a fix for you to test whenever I could.

crosswick, 2020-08-29:
Perhaps related? My setup is a late 2013 Macbook Pro running macOS Mojave 10.14.6:
https://github.com/tessel/node-usb/issues/277

hathach, 2021-03-06:
There is recent bug fix related to vendor control handling especially webusb https://github.com/hathach/tinyusb/pull/646 please pull and try again to see if it fix this issue

crosswick, 2021-03-06:
Thank you - unfortunately, still the same error

hathach, 2021-03-09:
hm, if you are not using the updated example, please also sync your project to the new webusb example handling for control transfer as well. https://github.com/hathach/tinyusb/pull/645

I just pulled out the board stm32f407disco and test with `webusb_serial` and everything works just fine with chrome on Linux. Please note that it is necessary to test with exact dev board and the exact example since that ensure we are testing the same thing.

crosswick, 2021-03-09:
I did use the updated example... also yes I have an exact
stm32f407disco board. I compiled in macOS and also in a fresh Ubuntu
20.04.1 VM and uploaded the resulting .elf file with STM32CubeProgrammer in
macOS, resulting in the same error in Chrome macOS. I will try again with
everything inside the Linux VM.

crosswick, 2021-03-09:
To doublecheck, the stm32f407disco board is this exact one, right? 
https://www.st.com/en/evaluation-tools/stm32f4discovery.html

hathach, 2021-03-09:
yes, it is that one, maybe it has something to do with macOS. There is a SET_INTERFACE targeted the webusb interface can you update the main.c `tud_vendor_control_xfer_cb()` of webusb serial to add following if to see if it could get it work

```C
else if (
        request->bmRequestType_bit.type == TUSB_REQ_TYPE_CLASS &&
        request->bRequest == 0x22) {
    // existing code
}
// new code
else if ( request->bmRequestType_bit.type == TUSB_REQ_TYPE_STANDARD && request->bRequest == TUSB_REQ_SET_INTERFACE)
{
    // response with status OK for SET_INTERFACE
    return tud_control_status(rhport, request);
}
```

crosswick, 2021-03-09:
Still the same unfortunately. Here's my updated code, for reference:

[image]

also, not sure if related but not long after this the wifi of my Macbook stopped working :) reboot hasn't resolved it

hathach, 2021-03-10:
@crosswick update: I also tested with one of my macos machine, it seems to work just fine. I guess you have to do the test here if you don't mind. Can you post the LOG with updated code to see if we got other error rather than the stall with SET_INTERFACE as previously. 

```
USBD Setup Received 01 0B 00 00 02 00 00 00
Set Interface
VENDOR control request
Stall EP0
```

crosswick, 2021-03-10:
Alright, I will look into it. Can you specify which version of macOS you are using, please? I've tested on Mojave, can also try on Catalina

hathach, 2021-03-10:
mine is 11.1 Big Sur, post the log up (with update code) when you have time tested with, we could analyze it more.

crosswick, 2021-03-10:
Well not sure why, but it works now... first I used my Catalina Mac Pro to compile and test - works. Same board also works on the earlier Mojave Macbook. Next, without changing the compile toolchain on that Macbook, I compiled the current version - also works. Perhaps my previous test of 4 days ago somehow wasn't with the updated version... 

I could doublecheck with a version prior to the vendor control handling fix if you'd like. In any case: thank you! Issue can be closed as fas as I'm concerned.

hathach, 2021-03-11:
thanks for confirmation, the #711 handling SET_INTERFACE at usbd even vendor cb does not handle it at application. If you would just pull and test with the latest master and it still works then I think we could close this for now. Maybe the make doesn't update the changes, best to try with a clean first :)

crosswick, 2021-03-12:
Yes, can confirm that it still works with #716

hathach, 2021-03-13:
Thanks for confirmation, to sum up it is lack of SET_INTERFACE handling, which only macos send occasionally. Just fix it at usbd level so that it is totally optional for class driver. Closed now

closed 2021-03-13 by hathach