hathach/tinyusb #501: How can I get  the signals of DTR and RTS flow control in real time？

stduinoiot, 2020-08-31:
Hello, I want to use The Webusb of TinyUSB to realize program download, which requires hardware DTR and RTS control. Currently, I can't find the function to get the streaming signals of DTR and RTS in real time by looking up the program by myself. I would like to ask which function can achieve the real-time acquisition of the streaming signals of DTR and RTS?

ladyada, 2020-08-31:
do you mean webserial? in that case you should look at the specifications from https://wicg.github.io/serial/

chegewara, 2020-08-31:
Maybe this:
https://github.com/hathach/tinyusb/blob/master/examples/device/webusb_serial/src/main.c#L169

stduinoiot, 2020-09-01:
> do you mean webserial? in that case you should look at the specifications from https://wicg.github.io/serial/
yes, webserial.
I have read the relevant information in this website provided by you, but still can not solve the DTR and RTS signal I want to get, still thank you for your prompt.

stduinoiot, 2020-09-01:
> Maybe this:
> https://github.com/hathach/tinyusb/blob/master/examples/device/webusb_serial/src/main.c#L169

I tried to get the DTR and RTS control signals using this interface function, but failed. Thanks again for the tip

hathach, 2020-09-01:
> > Maybe this:
> > https://github.com/hathach/tinyusb/blob/master/examples/device/webusb_serial/src/main.c#L169
> 
> I tried to get the DTR and RTS control signals using this interface function, but failed. Thanks again for the tip

there is no DTR/RTS equivalent for webusb, the pair process is between browser and OS. You need to implement the connect like the web example to send connect/disconnect event. You may want to look at webserial which use standard cdc instead.

stduinoiot, 2020-09-01:
> there is no DTR/RTS equivalent for webusb, the pair process is between browser and OS. You need to implement the connect like the web example to send connect/disconnect event. You may want to look [...]

So is there a way to get the DTR/RTS control signal in real time By Tinyusb? Can you tell me which interface function to use or how to get it?Looking forward to your reply. Thank you in advance.

cr1901, 2020-09-01:
So, now _I'm_ confused: webserial is a completely different protocol from "webusb protocol talking to a serial device" (like the `webusb_serial` example)?

duempel, 2020-09-04:
> So is there a way to get the DTR/RTS control signal in real time By Tinyusb's CDC?

Yes it is. In CDC class there is a callback `tud_cdc_line_state_cb` which is invoked, whenever a new line state is requested. You can take a look at cdc_msc exmaple. The callback is used there at line [L131](https://github.com/hathach/tinyusb/blob/10d5dac9137adfd6c14d96e1079f08447eacd9d8/examples/device/cdc_msc/src/main.c#L131) to print the welcome message when opening the port.

hathach, 2020-09-05:
> So, now _I'm_ confused: webserial is a completely different protocol from "webusb protocol talking to a serial device" (like the `webusb_serial` example)?

There is two web standard, the 
- webUSB is basically bind to any interface of an USB device, much like libusb claim interface. webUSB mostly used with vendor/custom class with either HID or BULK whichever the developer (device + web) like. `webusb_serial` is just an example that emulate the use of webusb like an virtual serial.
- webSerial is another std  that binds the actual serial port of the host PC such ttyACM0, ttyUSB0 and use the native serial driver to communicate with device. Here the device must implement an CDC class. In fact, browser is just another terminal client, device doesn't even know it is a browser. Somehow webSerial is much less popular, not widely adopted like webUSB. It is probably due to the security concern. 

Hope that help.

stduinoiot, 2020-09-07:
> Yes it is. In CDC class there is a callback `tud_cdc_line_state_cb` which is invoked, whenever a new line state is requested. You can take a look at cdc_msc exmaple. The callback is used there at [...]

Hello, I tried this interface function, but I can only get a signal once if there is data coming in. I am currently using the **tud_cdc_n_get_line_state** interface function for retrieving the DTR signal, but the RTS control signal is not yet retrieved. Thanks for the tip all the same!

duempel, 2020-09-10:
> I am currently using the tud_cdc_n_get_line_state interface function for retrieving the DTR signal, but the RTS control signal is not yet retrieved.

Which terminal software are you using to test this? I once spotted a similar behaviour when using HTerm. Changes to RTS were only received after changing DTR as well. I'd recommend you to test your implementation with other terminals as well. (Maybe some applications think it's not useful to set RTS if they are actually not ready 🤷‍♂️?) Putty for example sets DTR and RTS just after connecting to the port. If you get both set in your software the implementation is good. If not maybe you checked a wrong bit since `tud_cdc_n_get_line_state` returns a byte of information instead of the single bits you would get when using the callback.

stduinoiot, 2020-09-20:
> Which terminal software are you using to test this? I once spotted a similar behaviour when using HTerm. Changes to RTS were only received after changing DTR as well. I'd recommend you to test your [...]

Got it ,thank you very much.

closed 2020-09-20 by hathach